prusa3d/Prusa-Firmware #2114: Crashing into the bed during Z calibration

david-antos, 2019-08-16:
MK3 and/or MK3S, FW 3.7.2.

When Z calibration is initiated, the extruder is first lowered until it reaches the bed. If the extruder was it its rightmost position on X axes or near the rightmost position, PINDA probe doesn't reach the bed, being to the right of the bed. So PINDA doesn't trigger and the extruder crashes into the bed.

Reproducibility: always, behaves identically on 3.7.1 as well.

Possible remedies: homing might be a good idea before Z calibration.

leptun, 2019-08-16:
Can you please post a video of the issue from when you start the printer and with what commands you send to it?

I’m not sure which Z calibration you mean since there are more than one. Do you mean homing the Z axis individually after homing the other axis and moving the extruder to the right through gcode or lcd?

david-antos, 2019-08-18:
Well, it's a quite embarrassing, I cannot reproduce the issue any more. It seemed very deterministic but I cannot confirm the right sequence of events that led to such a behaviour. Standard "Calibrate Z" started correctly with homing after anything I tried.

I suppose closing the issue is fully appropriate now, I'll reopen if I learn more about it. Thanks for your time and apologies for the confusion.

closed 2019-08-18 by david-antos