prusa3d/Prusa-Firmware #2115: [3.8.0-RC1] - Printer reboot at the begin of the print (+PrusaSlicer 2.1.0-alpha1)

antimix, 2019-08-16:
I sliced the **[PRUSA Media control Volume Knob object](https://www.prusaprinters.org/prints/3592)** in MK3 MMU Single mode with PrusaSlice 2.1.0-alpha1

It was never printed. 
- The printer perform the Preheating
- The perform the 9 point calibration.
- Bring the head at home
- **The intro line was never printed and the filament was never loaded**
- the LCD blanks, and the printer reboot.

Note that the filament was never loaded, neither the MMU2 was moved.

**Possible cause.**

- The Prusa object was  "Body Abstract.3mf"
- The object contained embedded "vintage" slicer settings
- The produced code does not like to the firmware

**Temporary Solution**

- I changed the Printer Settings with the actual default PrusaSlicer 2.1.0-Alpha
- Resliced the object

It was printed.

Whatever the cause was (probably bad gcode) the printer should give a warning, and not a reboot.

**This is the offending code. It never went after the last line...**

```
; generated by PrusaSlicer 2.1.0-alpha1 on 2019-08-16 at 14:14:56
; 
; external perimeters extrusion width = 0.45mm
; perimeters extrusion width = 0.45mm
; infill extrusion width = 0.45mm
; solid infill extrusion width = 0.45mm
; top infill extrusion width = 0.40mm
; support material extrusion width = 0.35mm
; first layer extrusion width = 0.42mm

M73 P0 R159
M73 Q0 S160
M201 X1000 Y1000 Z1000 E5000 ; sets maximum accelerations, mm/sec^2
M203 X200 Y200 Z12 E120 ; sets maximum feedrates, mm/sec
M204 P1250 R1250 T1250 ; sets acceleration (P, T) and retract acceleration (R), mm/sec^2
M205 X8.00 Y8.00 Z0.40 E1.50 ; sets the jerk limits, mm/sec
M205 S0 T0 ; sets the minimum extruding and travel feed rate, mm/sec
M107
M115 U3.7.1 ; tell printer latest fw version
G90 ; use absolute coordinates
M83 ; extruder relative mode
M104 S205 ; set extruder temp
M140 S60 ; set bed temp
M190 S60 ; wait for bed temp
M109 S205 ; wait for extruder temp
G28 W ; home all without mesh bed level
G80 ; mesh bed leveling
G1 Y-3.0 F1000.0 ; go outside print area
G92 E0.0
G1 X60.0 E9.0 F1000.0 ; intro line
M73 Q0 S160
M73 P0 R159
```

michalxfanta, 2019-08-19:
We will try the scenario described and try to find a problem.

vintagepc, 2019-08-19:
^^ This can happen if the printer's filament sensor reports 1 and the FINDA reports 0. The printer will reset instead of printing the purge line.

antimix, 2019-08-19:
So, is a known firmware bug ?

vintagepc, 2019-08-19:
It can be if your IR sensor is stuck at 1... symptoms are a dead ringer. Not saying there isn't another possible cause though but your startup gcode looks normal. Is it reproducible if you switch back to the alpha slicer?

vintagepc, 2019-08-19:
Also... weird stuff occasionally happens at startup.. Most likely not related but you might be interested in #2094 and the discussion therein on printer buffer overflow/corruption at startup.

mkbel, 2019-08-20:
In gcode I see, that there is missing any kind of T command to select filament or prompt user for doing so. So this is known bug, that printer with MMU can not print gcode sliced for printer without MMU unless you don't use load to nozzle from menu before you start printing actual file.

gysnst2, 2019-09-09:
i get this issue randomly today 3 times in a row.  turn off the printer and walk away for a bit.  then everything's ok  but it is only the printer with the mmu2 attached

Mirarkitty, 2019-11-20:
This happened to me today with 3.8.1...

Mirarkitty, 2019-11-21:
gcode log from octoprint:

[reset_chatter.txt](https://github.com/prusa3d/Prusa-Firmware/files/3873413/reset_chatter.txt)

Recv: echo: 3.8.1-2869

It does not leave the reset loop by itself, I have no indication why it resets.

smurfix, 2019-12-03:
Could you please fix that? Rebooting without an error message is never a good idea.

leptun, 2019-12-03:
@Mirarkitty there appears to be a misconfiguration in your octoprint instance: Not queuing T3, that tool doesn't exist according to the printer profile or was reported as invalid by the firmware

Try manually sending T3 from the terminal from octoprint with printer preheated and report the result. If I am right, you’ll get the same error and the printer will not react.

Mirarkitty, 2020-01-11:
@leptun My setup in Octoprint is MK3 single nozzle single extruder, I don't think it made up the T3? I think that comes from PrusaSlicer. I'm not slicing with Octoprint and I'm not asking for tools in Octoprint.

leptun, 2020-01-12:
@Mirarkitty You must define the maximum number of filaments as the number of extruder. If you don't, then octoprint will filter out what it thinks are invalid extruder numbers and you will get `Not queuing T3, that tool doesn't exist according to the printer profile or was reported as invalid by the firmware` because according to your profile there is only 1 extruder, not 4 or more. Try changing it and see if the issue is solved. Also, have you tried using the SD card for the same gcode?

mkbel, 2020-01-13:
@leptun: There is not expected to be T3 command when gcode is generated in MMU single profile. It is expected Tx and Tc pair of commads to be there.

leptun, 2020-01-13:
@mkbel I was refering to the error @Mirarkitty provided in the file “reset_chatter.txt”. You can find the error I was talking about in the logs. This does not apply to the author’s problem. It is just an observation to what I saw in the log.

mkbel, 2020-01-13:
@leptun Ah, ok. But this means, that either file is not sliced in MK3 MMU Single mode, or Octoprint is replacing Tx, Tc or T? commands with T3 command.

Mirarkitty, 2020-01-14:
I had no idea octoprint was allowed to touch the uploaded gcode. This seems like a bug or misfeature?

closed 2023-10-05 by github-actions[bot]