prusa3d/Prusa-Firmware #2117: [3.8.0-RC] Settings/HW Setup/Texture1/First Layer Calibration go only to PLA temp

antimix, 2019-08-17:
I needed to set up the Textured sheet for PET-G.
I went in **"Settings/HW Setup/Texture1/First Layer Calibration"** 
The LCD asked for filament selection.
**Then it started pre-heathing process in PLA mode.**
No material temperature selection was asked.

jbinfl, 2019-08-18:
Antimix:  I think, from what I can tell with playing with this, the first layer calibration will always be PLA just like the old Live Z calibration before this was always PLA.  If you need to do a PETG or ABS or something with a different live z, then you have to use the 75x75MM calibration square process to figure what that is and then just add it in to the 2nd or 3rd sheet live z to get it to PETG.  

In short, the live z value seems to always have to be PLA and if you want a different filament you have to manually adjust that...  From what i can tell...

antimix, 2019-08-19:
Hi,
may be, but I expect a mature Firmware that will ask for the temperature
settings on the LCD, exactly like it happens when I select Load to nozzle.
ANY filament loaded to nozzle should have the correct nozzle temperature,
no exceptions.😉
And of course the first layer calibration should be available for all the
materials supported by the LCD material list.

vintagepc, 2019-08-20:
The gotcha is you can't select a sheet if it's uncalibrated. And if you don't print with PLA, you're kinda stuck. 
I did ask for an improvement, #2091, in that regard.

michalxfanta, 2019-08-23:
We prepare this feature in one of the next releases.

michalxfanta, 2019-10-14:
We implemented this feature in 3.8.1 RC1 firmware.

closed 2019-10-14 by michalxfanta