prusa3d/Prusa-Firmware #2118: Strange First Layer offset

antimix, 2019-08-17:
I use PLA on smooth plate, set at -1.620

I installed the Textured plate to print in PET. I forget to set it up in HW settings, and sterted to print.  So, at the beginning of the print I just to moved the Live Z- Offset  from -1.620 to -1.920

**- How is possible ? I would have expected to decrease the number since the textured plate in thicker.**

After the print, I went in HW Settings/Textured1/First Layer calibration (to store it)
But instead of storing, it started a first layer calibration process. 
OK. It has a bug where no temperature is asked (issue opened).

When the process started, the offset displayed on LCD was not -1.920, but 0

**In order to obtain the same thickness I had to move to -2.060 !!**

**- Why -1.920 had a complete different thickness ?**
**- There should be something wrong  on the distance measuring process. I was very worried that the firmware could  mess-up and scrap the plate.**

leptun, 2019-08-17:
It’s already been reported that there are some problems with varying first layer in 3.8.0-RC1. Keep this open though as other issues mentioned might be different.

DRracer, 2019-08-23:
@leptun can you please link the other first layer calibration issues like this here? Thanks

leptun, 2019-08-23:
@DRracer Working on it.

closed 2023-10-05 by github-actions[bot]