prusa3d/Prusa-Firmware #2119: Tune | Speed allows for impossible % values (printer still follows max print speed from g-code)

mavelli, 2019-08-18:
Background: 
i tried to find out how fast my mk3s can go and still produce usable results (assuming easy material and other physical constraints). so i compiled https://www.thingiverse.com/thing:277394 into g-code and executing it from the sd card as a basic setup and planned to "tune" the speed via tune | speed settings on the printer.

issue: 
print settings standard "0.20mm quality mk3" with standard values 
* 100 % (nothing "tuned") - 39 min - as expected
* 200 % (via Tune | Speed) - 20 min - as expected
* 400 % (via Tune | Speed) - calculated 9 min - print took 20 min as shown also correctly on the display, but it did not print at 4 x speed despite "tuned" to do so. this was not expected by me, as all other parameters accessible from the tune settings are representing real possible values (e.g. you cannot tune the nozzle temp to 800 °C)

expected behaviour: 
printer does not let user tune the speed above the % resembling max print speed set by g-code

alternative also possibly expected behaviour:
tune speed indicates the % at which it would have to override max print speed and effectively do so by users confirmation

mavelli, 2019-08-18:
[Ripple_test_0.2mm_PLA_MK3S_39m.gcode.txt](https://github.com/prusa3d/Prusa-Firmware/files/3513614/Ripple_test_0.2mm_PLA_MK3S_39m.gcode.txt)

mavelli, 2019-08-18:
sorry, completely forgot: 
MK3s on 3.7.2-2363

michalxfanta, 2019-08-19:
The answer is a little more complicated. Maximum movement speed is limited to 200mm / s. By default, in PrusaSlicer for a 0.15mm QUALITY print profile, the speeds are 45mm / s for the perimeter and 80mm / s for the infill, so it is also theoretically possible to raise the speed for the perimeter to 444% and the infill to 250%. Higher values ​​will no longer speed up printing. If you have a 200mm / s g-code, increasing the speed in the Tune menu will have no effect.

So, the maximum speed value in Tune depends on speeds in G-code.

mavelli, 2019-08-19:
i understand that - and how it is implemented now makes sense to me functionally (tune menu settings do not make the printer override max speed from gcode). my problem is not the behaviour of the printer but the interface - sorry to not be clear enough above.

my problem is with the tune speed dial possibility. you can rotate it to 999 % without the interface telling you that it will anyhow ignore it and use xxx % because that would fit the max speed limit provided by the gcode. 
all other tune settings resemble real life values and it is clear (e.g. overtemperature warnings) that the printer will not heat the nozzle to 800 °C but stop and use a lower temp because of constraints. 
it would be awesome to have the same also on the speed % dial - reflecting in the interface what the printer is actually doing in % over nominal speeds. or even better limiting the tune speed dial to this % based on the speed limits provided in gcode anyhow.

michalxfanta, 2019-08-22:
OK, I understand that. We will think about it.

closed 2019-08-22 by michalxfanta