prusa3d/Prusa-Firmware #2121: [3.8.0-RC1 MK3S] - automatic temp cal. goes on forever

RakNoel, 2019-08-19:
### Issue
After updating printer, before anything else, a temperature calibration was started. Knowing it would take a while I left the printer and heard the beep. When returning the printer was still doing step 6/6 and went on for about 30 min before i shut it down.

### HW
MK3S MMU2S

leptun, 2019-08-19:
Please try again and post a photo with the temperatures menu in the support menu after it gets stuck.

If the noctua fan starts spinning because of extruder temperature over 50c it might cool the pinda enough to not reach the target temperature.

It might also be that it doesn’t reach the temperature because of other factors such as a defective probe. That menu should tell us enough for now.

RakNoel, 2019-08-19:
I did not get a picture, and I'm printing right now. Although, I do remember the readings as I was watching them.

Extruder 50c/0c (Noctua on)
Heatbed 109c/110c (Ranging from 109-110).
Room temp 23c

Will post pic if it happens again

RakNoel, 2019-08-19:
Oh and "F?" if that is relevant

leptun, 2019-08-19:
Ummm... I don’t know what “F?” is. Can you explain it further.

RakNoel, 2019-08-19:
F? means that the MMU does not know its position as it has not yet been used

leptun, 2019-08-19:
Oh, ok then. It is not relevant. :)

boraca, 2019-08-24:
0 degrees means your hotend thermistor is disconnected or damaged.

leptun, 2019-08-24:
@boraca Ummm, no. The 0 deg is his hotend target, which is as it should.
I asked for the temperatures in order to see the pinda temp and the extruder temp. The thing is that when the extruder fan turns on at 50 deg, it actually also blows some cool air on the pinda preventing it from reaching the expected temperature.

This issue has been reported before, but I can’t find the other posts right now.

Sembazuru, 2019-09-19:
I've had similar issues with using M860 to wait for 35°C on cold days (bed is only 60°C) as part of my start gcode before mesh bed leveling. Tenting over the PINDA with a bit of paper towel helps insulate it from the room (and trap heat from the bed) to allow the PINDA to finish warming up. Maybe you are seeing something similar with your temperature calibration?

closed 2023-10-05 by github-actions[bot]