prusa3d/Prusa-Firmware #2125: Weird "Print from SD" menu behavior

leptun, 2019-08-20:
The video sums it up: https://photos.app.goo.gl/h4XR5QjQeRZzRU4s7
@DRracer Please try checking this issue on a random sd card.

DRracer, 2019-08-21:
@leptun yes, I've managed to simulate the same behavior ... there seems to be a +-1 offset issue. The Spitfire_.... filename should rotate, but the last character is sometimes blank (space) as if the filename wasn't long enough.
Are you willing to fix it ;) ?

maximlevitsky, 2019-08-21:
I don't thing I have seen exactly the same bug but I have seen something similar:
Sometimes long print file name just doesn't start scrolling until you nudge it with up/down encoder movement.

leptun, 2019-08-31:
@maximlevitsky The #2155 PR should fix that issue as well.

PR #2155 (2019-11-08): SD Scroll workaround
@DRracer This PR resolves #2125. It fixes the issue, but I don't really like the current SD filename scroll implementation. Might be worth looking into rewriting the feature.
closed 2019-11-08 by DRracer