prusa3d/Prusa-Firmware #2128: Building Large format MK3S

shawntexan, 2019-08-22:
So i am building a large format prusa style printer x 435, y 505, z345 i can get it to pass self test and see the new dimensions but it fails homing adjusted stallgaurg up still wont home correctly, it will move manaully fine but at slower homing speeds if thinks its at the home due to endstop sense.  as you can probabaly tell i am a noob at this.  please help as i have spent months on this and considerable amount of money trying to get it to work.

leptun, 2019-08-22:
You might need to change tmc2130_sg_thr_home and tmc2130_current_r_home values for X/Y in tmc2130.cpp

shawntexan, 2019-08-22:
ok i will look into that 
tmc2130_wr(axis, TMC2130_REG_COOLCONF, (((uint32_t)tmc2130_sg_thr_home[axis]) << 16)); do i go up or down

shawntexan, 2019-08-22:
and this i have the  super whopper e3d stepper motors 

//running currents for homing
uint8_t tmc2130_current_r_home[4] = {8, 10, 20, 18};

leptun, 2019-08-22:
`uint8_t tmc2130_current_r_home` increase the 8 and 10 a bit.
`uint8_t tmc2130_sg_thr_home` try increasing or decreasing the values shown there and see result. I'd suggest you tried setting up only one axis and going to the other after the first is working.

The values in the array have the axis in order:
X: first value
Y: second value
Z: third value
E: forth value

You might also see `TMC2130_SG_THRS_X`. These are changed in Configuration_prusa.h or in the variants folder depending on how you compile your firmware. Change these for crash detection during print.

shawntexan, 2019-08-22:
ok thanks i changed y to 20 it will now home y x is good also, how do i
change the value to match my build area for the xyz calibration i tried the
usual changes under configuration but it wouldn't move to the larger
coordinates.  thanks again

leptun, 2019-08-22:
XYZ calibration or Mesh bed leveling?

XYZ calibration is only possible with special points in the bed.
Mesh bed leveling can be used as long as PINDA triggers on your bed surface.

shawntexan, 2019-08-22:
ok mesh bed leveling then i have pinda and aluminum build plate.

looks like y isnt moving back far enough for home i set it to -30 and -50
but doesnt seem to be moving to that position #define MANUAL_Y_HOME_POS -30

leptun, 2019-08-22:
MANUAL_Y_HOME_POS isn't actually used for anything... It's just there for who knows what reason from a long time ago

shawntexan, 2019-08-22:
ok i will print some hard stops then so it aligns with the bed

shawntexan, 2019-08-22:
should i be able to run z calibration?

shawntexan, 2019-08-22:
or more simply what stuff will i not be able to do that is too specific to the prusa designs, i have pinda probe alluminum build plate, no steel sheet though im hoping to have it as prusa automatic as i can get.  thanks again

leptun, 2019-08-22:
If you have stoppers at the top of the Z axis you should be able to use it. Just make sure you update the max positions accordingly for your setup since your printer is bigger.

Other than XYZ calibration, everything should work

shawntexan, 2019-08-22:
ok i will look at it some more tomorrow i am getting tired i know the z calibration is error out on the way back down but i will check into it tomorrow,  thanks for your help im sure i will need a lot more once i get to the calibration stuff but now that i have the axis moving i can do more.

leptun, 2019-08-22:
Please connect to the printer over usb and the serial monitor from arduino and post what the error says.

shawntexan, 2019-08-23:
here is my error when try to run z calibration
Recv: tmc2130_home_enter(axes_mask=0x04)
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: tmc2130_home_exit tmc2130_sg_homing_axes_mask=0x04

shawntexan, 2019-08-23:
Fixed my y axis so now home stops over the build plate

shawntexan, 2019-09-02:
back, got the new hardware designed but need to change the z calibration to match my new larger bed size been trying to print basic paterns but level is off hoping z will help also cant get the z calibration to work just goes down then stops.  would be happy if the z level worked before each print

closed 2023-10-05 by github-actions[bot]