prusa3d/Prusa-Firmware #2130: Printer fails to print file with special characters (^) in name

jokohoko, 2019-08-22:
This bug annoys me mostly because Solidworks for excample often saves files with ^ in the filename when working in assemblies.
I have not gone trough a list of characters to make an overview but i guess a solution would either be for this specific character or for all special characters at once.
The printer is able to display the name correctly. But he states the file is incomplete upon starting the print. (I am certain the file is complete and after changing the name it prints fine).

I also reported this to the PrusaSlicer issues page since another solution is to block the software from saving with this name.

leptun, 2019-08-23:
The thing about ^ that breaks the firmware is that ^ has special meaning in GCODE. It is used to delimit the command from the checksum. I don't think there can be a workaround around this issue since that is the standard.

jokohoko, 2019-08-23:
I agree, therefore i also posted the issue to the [PrusaSlicer](https://github.com/prusa3d/PrusaSlicer/issues/2800) page in order to maybe have a fix there,

DRracer, 2019-08-27:
@jokohoko thanks for referencing this issue from the slicer too ;)

@leptun can you please explain in more detail?
I was looking into the code a few hours ago and actually made a fix here:
https://github.com/prusa3d/Prusa-Firmware/pull/2137

DRracer, 2019-08-27:
@leptun thanks for your confirmation. There is an explicit code checking for the '*'. 
However, if you look at my PR, the '^' processing looked more like a forgotten debug something...

leptun, 2019-08-27:
@DRracer I might be mistaken about ^ being the character used. I think it's actually * that's being used.

When a command is sent from let's say octoprint, instead of being `G1 X100 Y50 F3000`, it's actually `Nxxxx G1 X100 Y50 F3000*yy`, where x is the line number and y is the checksum for the line.

real scenario: `N0 M106*36` 
This means that the filename can't contain *, not ^

mumboz, 2020-01-18:
I just experienced the same problem, and wanted to add to the report, that it was with the danish character 'ø' in the filename. Renaming the file resolved it.

closed 2023-10-05 by github-actions[bot]