prusa3d/Prusa-Firmware #2132: Spool Join After Runout Printing in Air with No Filament Loaded

TheDuggem, 2019-08-23:
I started an overnight print to use the remaining filament on one spool and automatically finish with a new spool using Spool Join.  Instead, the printer used the remaining filament on the first spool, failed to load the new spool, and was printing in mid-air with no filament loaded.  Unfortunately I wasn't near the printer to see exactly how it failed.  I didn't think to check sensor status on the display but the LED on my modified IR sensor was not on and the FINDA LED was not on.

i3 Mk3S v3.7.2
MMU2S v1.0.5
PrusaSlicer v2.1.0-alpha1
Printing in single mode.

Partial spool loaded in position 1.
Full spool loaded in position 2.
Filament sensor enabled.
Spool Join on.

Selected filament 1 for start of print.
Printed remainder of first spool.

Failed to load second spool.
Printing in mid-air with no filament loaded.
Position 2 red LED on MMU2S slow flashing.
Both filament sensor hardware were inactive.

This doesn't seem like a model / slicer issue but I will post a 3MF file if requested.

TheDuggem, 2019-08-29:
Possibly related to #1770

gysnst2, 2019-09-09:
ive tried to use the filament join but im my cases the filament triggered the finda and then i get mmu requires you attention.  it turns out that the extruder tried to eject a small peice of filament the size of the ptfe tube and the filament then gets stuck. and the MMU thinks it should wait for me to pull that piece out and then hit the mmu button to continue

TheDuggem, 2020-01-27:
I attempted a spool join again over the weekend.  I happened to be present for the filament change.  

The spool was empty enough that I was able to cut the 90 degree bend off of the end of the filament but it still had a tight curl which caused the filament change to fail.  Mechanical issues aside, the system does not seem to detect the resulting error condition and continues on.

The old filament could not be unloaded through the MMU
The MMU detected a load error
The FINDA was not detecting filament
The LED for the new spool was flashing red slowly
The printer continued printing in spite of the MMU error
No MMU message appeared on the printer display

I have more notes on the sequence of what happened if you want them.

Hopefully this is useful.

closed 2023-10-05 by github-actions[bot]