prusa3d/Prusa-Firmware #2138: Hint: The continuation arrow in the new "Sheet"-menue entry should be removed.

TOTO-G, 2019-08-27:
Clicking at this menue entry toggles the defined sheets but does not jump to an other menue. Therefore the arrow should be removed.
Would it be possible to recognize a long press of the knob at this menue line and jump direct to the hw sheet setup menue?

michalxfanta, 2019-08-29:
That's a good point, we'll fix it, thanks.

FloatingCameras, 2019-08-30:
When I first updated I was presented with Smooth for the Plate. The arrow was an indicator I should click, but **when I clicked nothing changed**, the Plate remained on Smooth.  Checking Settings I found HW Setup.  I didn't go through a Calibration so I just renamed the Plates.  After I calibrate the Smooth sheet clicking on Sheet would toggle between the two. I renamed the third one Unused go make it clear it wasn't in use.  

Without the arrow I don't know what would prompt a person to click to change the value.  

The first entry should be Textur1 since I expect most people are using the textured sheet and that way it matches if they upgrade and do nothing else.  

It wasn't clear to me that I had to calibrate the sheet to use it but that does explain the Reset option.

Just some input based on my experience.

A very nice feature.

leptun, 2019-08-30:
An update has been merged today. The sheet selection on the main menu should only appear when you have at least two initialized sheets.
The changes should appear in the next release.

FloatingCameras, 2019-09-02:
MK3S/MMU2S -- 3.8.0-RC2 Build 2639  - Self Compile: Verification of Sheet Menu changes.

1) Reset all but one filament -> Sheet menu item still showed, which I think is fine. Clicking didn't change the selection since there was only one defined.  I feel the arrow is an indicator there is something that can be changed even when it doesn't change anything.
2) Added a second sheet, Settings->HW Setup. Since I knew the Z Adjust value I started  the Layer Calibration on the second sheet and instead of selecting a filament I selected Cancel, which leaves the steel plate selected, and then I was able to set the 'Live Adjust Z' value manually.
3) Now when the menu item "Sheet" is selected and the button is clicked it toggles between the two defined sheets, which is as I expected.

Everything functions as I would expect. I will enjoy this  new feature.
👍

michalxfanta, 2019-10-07:
This is solved in 3.8.0, closing ...

closed 2019-10-07 by michalxfanta