prusa3d/Prusa-Firmware #2144: MK3 MMU2S Feeds wrong filament

rackley096794, 2019-08-29:
MK3S: 3.7.2
MMU2: 1.0.6

I'm seeing a consistent issue and there is clearly at least one bug here, possibly two. When first powered up the MMU2S usually thinks it is on #1 but sometimes thinks it is on #4.  Currently I only print off of #2 (single filament) so it is always physically on #2 when it powers up.  Because I'm doing single filament prints only, it always asks me which filament to print at the beginning of a print.  I enter #2 in the keypad and it indicates ?>2 which I believe means it is trying to move to #2?  However, the MMU2S LED still indicates it is on #1 and of course it tries to feed #1 filament (which is empty) down the PTFE tube to the extruder.  

So it seems there may be two different issues here..  #1 is that the MMU2S starts off with a disparity between the perceived and actual position of the filament selector (the slider, not the barrel), and #2 is that the MK3 and the MMU2 aren't communicating the filament change to #2 as the LED on the MMU2S still indicates the MMU2S thinks it should feed #1 (which it does, and is incorrect).

safetychief8, 2019-08-31:
With mine, the same issue, but the green light is on F1, but the selector goes to F2, also, twice I had to reset the MMU2S, because 2x in the current print I'm doing, all 5 red and green lights start flashing and I have to hit the reset button to resume printing. I informed Prusa about the issue, hopefully there'll be a fix soon, going back to 1.0.5 in the meantime.

rackley096794, 2019-08-31:
I will probably go back to 1.0.5 as well and maybe drop MK3S firmware revs as well.  The current versions are so broken it sets a new standard....even by MMU standards.  :-(

closed 2023-10-05 by github-actions[bot]