prusa3d/Prusa-Firmware #2151: Nozzle and bed do not turn off at end of print MMU2S MK3

rackley096794, 2019-08-30:
MK3S: 3.7.2
MMU2: 1.0.6
PrusaSlicer 2.1.0 Alpha1

Two problems here:

1)  At end of a single-filament print, the bed and nozzle do NOT turn off.  

Gcode at end of print:
```
G1 E-10.0000 F3100
G1 E10.0000 F3150
G1 E-10.0000 F3250
G1 E10.0000 F3300

M702 C

G4 ; wait
M104 S0 ; turn off temperature
M140 S0 ; turn off heatbed
M107 ; turn off fan
G1 X0 Y200 F3000 ; home X axis
M84 ; disable motors
M73 P100 R0
M73 Q100 S0
```
[image]

2)  When the temp timer expires, the filament needs to be retracted first.  As it works right now, the printer will turn off bed and nozzle temp due to timer.  However, the nozzle and bed cool down FIRST, then the user clicks the OK prompt on the control wheel, THEN the filament tries to retract/unload.  Of course now the filament is cold and solid.

gudnimg, 2022-12-10:
@Prusa-Support @rackley096794 I believe this ticket is obsolete by now. Please consider closing the Github issue.

I've never seen this issue before. I got my printer after 3.9 was released so maybe this was fixed between 3.7.2 and 3.9

Prusa-Support, 2022-12-22:
Hello @rackley096794, sorry that we didn't respond regarding this issue back then. It should be indeed fixed by now, I'll close the issue. 

Alessandro Pantaleo
Prusa Research

closed 2022-12-22 by Prusa-Support