prusa3d/Prusa-Firmware #2152: [3.8.0-RC2] - Strange MMU needs user attention at the end of printing

antimix, 2019-08-30:
Yesterday after a 5 hour print I found the i3MK3S+MMU2S in this strange status.
You can see here in the video: [https://www.youtube.com/watch?v=8gTh_rbjSys](https://www.youtube.com/watch?v=8gTh_rbjSys)
The print seemed finished, the nozzle was still hot at 215°, filament was unloaded, and ALL MMU2 leds flashed green+red at the same time. 

- Pressing any MMU2 button did noy had any effect.
- Pressing the selector button on MK3 did not had any effect.

**None of the safety nozzle turn off had worked.**
I removed the filament pulling the spool (at the beginning, not in the video), and I feel that it was completely and correctly unloaded.  You see here , that even opening the MMU2 shows that there is not any jam or filament inside. However either the MMU2S either the MK3S are blocked, and surprise, turning off and on, "automagically" fix any issue (so, no real jam).
I don't know if this was caused by the latest MK3S firmware or MMU2 firmware.

Today after I retried another 11 hours print all was fine.
It is clear that there is still a lot to work in the interactions between Marlin and the MMU2 firmware.

leptun, 2019-09-08:
Your FINDA led seems to have indicated that there was filament passing the selector. I've also encountered this, where the ball would not fall with gravity and still trigger the sensor. The solution was clearing the ball hole and re-positioning the FINDA. All MMU2 leds flashing on startup means that Filament is detected by the Selector.

The automagically part probably happened by the vibrations from the idler homing making the ball fall and getting the FINDA in the correct state. That's my theory.

antimix, 2019-09-08:
Hello, too good to be true....

unfortunately is not the cause. In another similar case I had, as you can
see in the video, I decided to verify the ball movement, AND dismount the
FINDA,  just to discover that it "feels" object in the air....
I verify with PRUSA that in reality FINDA and MMU2 board is is OK, it is a
bug in the firmware that does not update the MK3 status.
Still to be fixed I think.
https://www.youtube.com/watch?v=U6xqnDquI1g&t=3s

gysnst2, 2019-09-09:
ive had this issue come up a few times it was strange to me

gudnimg, 2023-07-31:
Closing as this is now resolved with FW 3.13.0 and MMU FW 3.0.0 :)

closed 2023-07-31 by gudnimg