prusa3d/Prusa-Firmware #2156: Print stops in the middle

kyjak, 2019-08-31:
Slicer - CURA
Connection - Octorprint, direct SDcard print
The printer just stops at height 43.95, no message, no power interrupt, no resume option.
Just crashes. Tried both octoprint and direct sdcard print, exactly same behavior.
There is nothing wrong with the gcode, checked the code before and after that Z height.
Attached is the [code.[](url)](url

)

leptun, 2019-08-31:
No gcode is attached

kyjak, 2019-08-31:
Probably the file was too large. However, closing this one. Downgraded FW to 3.7.2 and seems to work.
3.8.0 RC2 has probably this bug in it.

closed 2019-08-31 by kyjak

leptun, 2019-08-31:
Please reopen the issue. If what is happening is not a hardware issue or some old obscure bug in the firmware, then it is a serious issue that has to be fixed before 3.8 final is released. Please post more details so that we can figure out the issue.

Is the issue repeatable?
Please post the gcode as a zip file.
You said you were using octoprint. Try printing without it connected.
Also, in what way did the firmware crash? (what could you do at that point in the firmware?)

kyjak, 2019-09-01:
Ok, here you go. Reattached as a zip file. Hope works this time.
The issue is gone by downgrading to 3.7.2.
The printer did stop simply the job without recovery option. 
Two times at around 44mm, one time lower, did not measure.

[OPI3MMK3_noha specialna 70mm BR.zip](https://github.com/prusa3d/Prusa-Firmware/files/3563702/OPI3MMK3_noha.specialna.70mm.BR.zip)