prusa3d/Prusa-Firmware #2157: Digital signature is invalid

kyjak, 2019-08-31:
Users cannot open the COM port of prusa cause of missing digital signature under windows 10.
[image]

Janys, 2019-09-02:
Hi! Are you using the latest drivers 2.2.6 available from https://www.prusa3d.com/drivers/ ? We have recently updated the package with newly signed drivers. The signature should be valid till March 2020.

kyjak, 2019-09-02:
Yes, I uninstalled all prusa, including drivers, installed 2.2.6. I think the old drivers got stuck somewhere, any steps for "forced" total removal?

Janys, 2019-09-02:
Yes, right-click the faulty driver in the Device Manager and select Uninstall, then tick the box saying "Delete driver software for this device". That should ensure the drivers are removed completely. Then try reinstalling 2.2.6. Please let me know if it worked. I'll definitely do some change to the code, so the next driver package will remove these old drivers automatically.

closed 2023-09-27 by 3d-gussner