prusa3d/Prusa-Firmware #2158: Strange Layer Shift on 3.8.0 RC1

crashtest22, 2019-08-31:
I had 1 print do a strange layer shift and then corrected about a 20 layers later. was on the x axis. I found no issues with other prints. 
I have now flashed to RC2 and will report any issues.
Printing over octoprint sliced with simplify3d

leptun, 2019-08-31:
Things to check:
Check the hardware for loose pulley. It could shift back to the initial position if the screw is slightly loose.

Check if crash detection was turned on.

Check if the model is too big and the X axis might have collided with the axis end. Happened to me before.

Do a selftest as to calibrate the auto-home for X and Y. If you don’t do so, if a crash occurs and is detected it might actually home inconsistently and actually cause the layer shift.

crashtest22, 2019-08-31:
I did check all pulley's and belts.
All tight and the spec on belt status was in range

Crash detection was on but print was middle of bed and not large
very far from edges. 
I had done self test after print to get belt status and have not had it happen again.
currently printing attached gcode. 
Will report here if it shifts 
[dx9BattCover_1800mahb.zip](https://github.com/prusa3d/Prusa-Firmware/files/3562697/dx9BattCover_1800mahb.zip)

crashtest22, 2019-08-31:
Here is a few pics and Gcode of the print that shifted
[image]
[image]
[Groot_v1_1M_Merged.zip](https://github.com/prusa3d/Prusa-Firmware/files/3562705/Groot_v1_1M_Merged.zip)

william14235, 2019-09-11:
Mine shifted once too but it only happened once. However it was a small print so I don't think the size has to do with it

Prusa-Support, 2022-06-28:
This issue was not proved to be concerning the firmware.

Apparently, this issue was not reproducible so it was likely a sporadic sequence of crashes or layer-shifts.
The printer may have detected some crashes along the axis and may have home incorrectly for some reason (similarly to issue #1096). It may have crashed again later and homed correctly.
We cannot even exclude that two layer-shifts occurred one after another in opposite directions, where "almost coincidentally" the second one corrected the print position, as mentioned above in the thread.

This issue will be closed because stale and not reproducible.

Michele Moramarco
Prusa Research

closed 2022-06-28 by Prusa-Support