prusa3d/Prusa-Firmware #2159: Wrong filament after MMU power fault (3.8.0-RC2, 1.0.6)

vintagepc, 2019-08-31:
Started getting MMU power faults (all 5 MMU LEDs blinking red and green). Instructions say to reset the MMU to correct. However, the MMU will re-home and park on filament 1, not the correct filament the T command had requested. It can be worked around if you're quick and can hit "change filament", but not ideal.

MB3Y, 2020-02-09:
I'm too noob in GitHub, but know how to submit a problem. The simple access to Prusa GitHub was PrusaSlic3r-Help-Report a problem... so my submission ended up here for the moment https://github.com/prusa3d/PrusaSlicer/issues/3628
problem #2159 seems related to my finding.
May I ask if this problem is considered to be taken up any time soon? Original submission dates 6months back meanwhile and before I submitted I was checking for SW updates... bad luck however

MB3Y, 2020-02-09:
Resubmitted in the correct GitHub repository under #2476

JCaba, 2020-03-12:
I just had this same problem today:  I resolved as follows (maybe ulimately related to "spooljoin" during reset?

March 12 2020:
Was using MMU positions/filaments stations 1 and 3:
MMU indicated line 3 (soluble) was jammed, when in fact line 1 (PLA) was jammed in the print-head.  After extracting line 1 (by faking the machine into thinking line 1 was clear (by removing front PTFE tube and clearing the MMU, then pause MK3, remove side fan, increase head temp to 250, then extracting filament from print-head).
Then after resume, the MMU wanted to load line 5 – which wasn’t being used and was actually empty. (I faked it – by temporarily inserting filament so it was ‘loaded’) , then it wanted to load line 2 (faked it again).  Printer started to use line 2.
Paused, went to settings, turned off the ‘spooljoin’ function, reset MMU, Then the MMU/Mk3 resumed using the normal/selected filaments (line 1 and line 3).

closed 2023-10-10 by github-actions[bot]