prusa3d/Prusa-Firmware #2160: Display issue when filename has exact 8.3 format

DonGiotto, 2019-08-31:
As we know, Marlin lists gcode file from sd in the old DOS 8.3 format. Newer marlin versions should support the M33 command for returning the "original" filename insted of the short version.

For a little project of mine, I use Octoprint to report all prints with events to a database. And because the Prusa Firmware doesn't support this command, I decided to use "real" 8.3 file formats with a simple format like MMDDYYNN.gco.

For testing purposes I added following files on the SD card:

**With gcode extension:**
fail.gcode (8 or less)
failfail.gcode (exact 8)
failfail1.gcode (more than 8)

**With gco exention:**
fail.gco (8 or less)
failfail.gco (exact 8.3 format)
failfail1.gco (more than 8)

In the menu on the display this happens:
**With gcode extension:**
- fail.gcode    -> OK
- failfail.gcode      -> OK
- failfail1.gcode  -> OK

**With gco exention:**
- fail.gco - displays as FAIL.GCO and disapears when selected with the scroll wheel
- failfail.gco - displays as FAILFAIL.GCO and disapears when selected with the scroll wheel
- failfail1.gco -> everything fine as it should be

leptun, 2019-09-02:
I can confirm this issue. I've had this happen when uploading files from octoprint to SD and trying to print them from the lcd afterwards. Also ghost files would exist.

DavidUnboxed, 2019-09-11:
This combined with the inability of Octoprint to support longer than 8.3 makes file names very challenging.  I suggested on Octoprint just this past week that using a table to store 8.3 and long name format but always writing the short name might be useful.  :(

leptun, 2019-09-11:
I’ll look into the issue this week. It appears as if the scrolling filename treats all files as LFN and reads a blank name since it only has a short name.

.gcode forces LFN
9+ name forces LFN

If the above conditions are not met (it is SFN), the filesystem is not required to write a LFN to that file, hence why it’s blank.

DonGiotto, 2019-09-11:
> 
> 
> This combined with the inability of Octoprint to support longer than 8.3 makes file names very challenging. I suggested on Octoprint just this past week that using a table to store 8.3 and long name format but always writing the short name might be useful. :(

imo off-topic, but such a table will not work with the Prusa Firmware, as the M33 command is not implemented (yet). So neither Octoprint nor you will get any LFN over the serial connection.

DavidUnboxed, 2019-09-11:
No. I think you are right on topic. Except: The side table would store 8.3 formatted primary keys and contain the LFN and any other details on the Octoprint side leaving the printer to only need to deal with 8.3. This would make it less friendly on the console display of the printer. However, would Mike just play an Octoprint better. Just call the 8.3 formatted names 'job ID numbers' and ibecomes a security feature for some institutions.

leptun, 2019-09-11:
@DonGiotto a fix has been proposed in #2195.

Sembazuru, 2019-09-19:
Is this also the cause of trying to start a SD card print from pronterface failing? I've only ever tried long filenames, I haven't put the time into playing with differently sized filenames. But, I would assume both Octoprint and Pronterface use similar (if not the same) gcode commands to initiate SD card prints.

leptun, 2019-09-29:
@DavidUnboxed @DonGiotto Take a look at this:
https://photos.app.goo.gl/xVqXAvABzW23c3Q26
I'm currently polishing the feature, but it looks very interesting and it works!

leptun, 2019-09-29:
@Sembazuru Open file in pronterface fails because sends the wrong filename back.
it sends `M23 wii.gco 7720` instead of `M23 wii.gco` where 7720 is the filesize. This is reminiscent from when Prusa-Firmware split from Marlin a long time ago. This has changed in marlin, but not also in Prusa-Firmware.

DonGiotto, 2019-09-29:
> @DavidUnboxed @DonGiotto Take a look at this:
> https://photos.app.goo.gl/xVqXAvABzW23c3Q26
> I'm currently polishing the feature, but it looks very interesting and it works!

Very nice! How do you get the full file name with octoprint? I'm still looking for a way, to get the full name with the octoprint event feature.

leptun, 2019-09-29:
I using one of octoprint's built-in plugins called Action Command Prompt.
Read more here: http://docs.octoprint.org/en/master/bundledplugins/action_command_prompt.html#sec-bundledplugins-action-command-prompt

All changes are in the firmware. No changes on octoprint's side are required (except for a plugin for that `Print form SD` button. It just sends `M876`).

Example communication:
```
Send: M876
Recv: //action:prompt_begin Print from SD
Recv: //action:prompt_choice Close Menu
Recv: //action:prompt_choice [..]
Recv: //action:prompt_choice [SPOOL]
Recv: //action:prompt_choice [MMU]
Recv: //action:prompt_choice Lattice_Cube_by_LazerLord10_0.15mm_PLA_MK3SMMU2S_2h1
Recv: //action:prompt_choice MK3_PRUSA_dual_color.gcode
Recv: //action:prompt_choice MK3_PLA_Prusa_200um_20M.gcode
Recv: //action:prompt_choice MK3_PLA_Buddy_150um_2H.gcode
Recv: //action:prompt_choice MK3_PLA_Vase_200um_6H_25M.gcode
Recv: //action:prompt_choice MK3_PLA_Triceratops_Skull_150um_5H.gcode
Recv: //action:prompt_choice MK3_PLA_Treefrog_150um_variable_1H_25M.gcode
Recv: //action:prompt_choice MK3_PLA_Treefrog_50um_3H_40M.gcode
Recv: //action:prompt_choice MK3_PLA_PrusaBeerOpener_ColorPrint_200um_50M.gcode
Recv: //action:prompt_choice MK3_PLA_Nefertiti_150um_7H.gcode
Recv: //action:prompt_choice MK3_PLA_Gear_Bearing_150um_2H.gcode
Recv: //action:prompt_choice MK3_PLA_Castle_100um_12H_30M.gcode
Recv: //action:prompt_choice MK3_PLA_Batman_200um_18M.gcode
Recv: //action:prompt_choice MK3_PLA_Adalinda_200um_7H_30M.gcode
Recv: //action:prompt_choice MK3_PLA_3DHubs_Marvin_100um_1H_7M.gcode
Recv: //action:prompt_choice MK3_PLA_3DBenchy_150um_2H.gcode
Recv: //action:prompt_choice MK3_PLA_Whistle_200um_30M.gcode
Recv: //action:prompt_choice WII.GCO
Recv: //action:prompt_choice TEST.GCO
Recv: //action:prompt_show
Recv: ok
```

As for starting the print, it's simply M876 S<id>, where <id> is the item selected (0-based index).
eg `M876 S3` for [MMU] folder.

leptun, 2019-09-29:
You might have noticed the layout error with long text in buttons. I've submitted an issue in Octoprint's github. I hope they fix it soon as this looks like a great feature that might make using FlashAir super easy entirely from Octoprint.

leptun, 2019-09-29:
Also there appears to be a character limit to the filename (53). It's the same if it's scrolling on the lcd so there is nothing for me to do there. See `Lattice_Cube_by_LazerLord10_0.15mm_PLA_MK3SMMU2S_2h1`, the filename is incomplete.

3d-gussner, 2019-09-29:
@leptun I remember that some mentioned that there is limit how many characters you can submit via gcode in one line. Could that also be a a reason to for the limit of filename lengths?

leptun, 2019-09-29:
@3d-gussner It is not the case here, since in the firmware itself the LFN is truncated. It is correct what you are saying that there is a limit to the length of a command, but that is 96 chars, almost double of this filename.

closed 2019-11-08 by DRracer

krisrok, 2020-06-07:
@leptun the "flashair" solution above is just the right ratio of hacky and working :) do you have it in a repo somewhere for others to try? i tried but could not find it in your account.
or another place to discuss it instead of this kind-of-related issue?

leptun, 2020-06-07:
I did not use it in quite a while, but here is the branch. https://github.com/leptun/Prusa-Firmware/tree/MK3_OP_LCD

krisrok, 2020-06-07:
great, thanks! is there another alternative for flashair/printing from sd or didn‘t you follow the approach any further?

PR #2195 (2019-11-08): Fix short filenames scrolling in sdprint menu and status screen
@DRracer Fixes #2160