prusa3d/Prusa-Firmware #2161: Feature Request: Rehome X/Y in tune menu when paused

vintagepc, 2019-08-31:
I'd like to request a "tune" menu item to re-home X and Y. I've found myself needing/wanting to do this in a number of cases, either when having paused a print to address something like warp/lifting or filament change with an unintended bump of the X/Y causing a layer shift. 

I can think of any number of other cases where it might be nice to re-home X and Y on demand mid print after addressing an issue, which could save a several hour print from becoming a spaghetti blob.

Not a feature that will be used particularly often, but a lifesaver when it is actually needed, since connecting to USB resets the printer.

michalxfanta, 2019-09-19:
Now, when printing is paused, the extruder moves to the left-rear position, how much is it different from the left-front position, which you need?

vintagepc, 2019-09-19:
That doesn't re-home the printer's zero position, does it? So if there's a minor layer shift it will still persist. But if you can rehome X/Y when paused it can be easily corrected.

3d-gussner, 2019-09-19:
@michalprusa I think it is not the pause parking position is an issue. But WE clumsy users may unintentionally hit the bed or the extruder and then a Rehome xy funtcion in Tune menu would prevent a failed print due to US clumsy users forced a layer shift.

vintagepc, 2019-09-19:
^^ Yeah, or there have been a few times where I've been sitting next to the printer and heard it thump and skip a step from hitting a print booger; I can pause it right away, clip the booger, and (ideally) rehome and continue on as normal; if I'd left it alone it would have introduced a layer shift; now I'd only have a small imperfection that can be more easily trimmed or filled.

michalxfanta, 2019-09-19:
OK, I finally understand that.

3d-gussner, 2019-09-19:
> ^^ Yeah, or there have been a few times where I've been sitting next to the printer and heard it thump and skip a step from hitting a print booger; I can pause it right away, clip the booger, and [...]

Another good example where a Rehome xy in Tune menu could save our prints .... Last Night a User saved my Print...lalala...He cut the bugger and Rehomed xy ... lalala ... Last Night a User saved my Print 

Glad I am not a song writer :-)

michalxfanta, 2019-10-30:
@vintagepc please contact me at michal.fanta@prusa3d.cz thx

vintagepc, 2019-11-06:
I have a tentative implementation in vintagepc/Prusa-Firmware/blob/2161-tune-rehome-X-Y/. 
Still need to test it but will file a PR when ready.

leptun, 2019-11-06:
@vintagepc Those changes only help if you detect the crash. If you don’t, then it doesn’t really help.

There might be problems with mesh bed leveling, skew correction and live Z if you use G28 as the rehoming command. I suggest that you try simulating a detected crash since that function already does the rehoming part correctly. @wavexx is also making some improvements to it, so that should help.

vintagepc, 2019-11-06:
> @vintagepc Those changes only help if you detect the crash. If you don’t, then it doesn’t really help.
Correct, this is the intent. I am usually in the same room as the printer and catch this sort of thing pretty quickly, so this would be a nice QoL improvement. 

> There might be problems with mesh bed leveling, skew correction and live Z if you use G28 as the rehoming command. I suggest that you try simulating a detected crash since that function already does the rehoming part correctly. @wavexx is also making some improvements to it, so that should help.

Thanks for the pointers. I'll look into that. Though if the crash detection becomes more reliable then you are right that this isn't really necessary anymore.

leptun, 2019-11-06:
@vintagepc He is fixing some problems related to pausing and resuming a print. This encompasses Crash Detection, Power Panic, Filament runout / recheck(old Fsensor). In essence all pause/resume actions that instantly stop the printer movement and do some actions.
I never said it isn't necessary. In fact I've also implemented something similar for my MK2 back when it was "original" that performed exactly that. It wasn't the best implementation, but it worked just fine. (fw. ~3.3.x).

vintagepc, 2019-11-06:
Ah, sorry, I misunderstood. I read your first post as pending improvements to crash detection and inferred from there.

vintagepc, 2019-11-14:
I finally had time to get back to this. I believe I see what you were referring to w/r/t things like saving the print state before rehoming, etc. 
It seems like the condition of pausing the print already takes care of that though; 

But, this raises an interesting point as to the availability of this function - while printing or while paused. 

I think there are cases for both options, so I'm thinking to implement as follows:

-> Place feature in the "in a print" menu so it's available when printing or paused

when selected:
-> if printing, rehome on-the-fly with a simulated crash by use of crashdet_stop_and_save_print, gcode_g28, etc, etc, being careful not to increment the crash counters.

-> if paused, print state is already saved, so just use gcode_g28. 

Thoughts/comments/concerns?

leptun, 2019-11-14:
@vintagepc I’ll have to look into this more indepth. This is most likely to happen during the weekend.

vintagepc, 2019-11-14:
Understood, thanks. I'll get the framework in place and await any advice.

ulope, 2020-12-14:
Until there's a proper solution to this there is a (somewhat brutal) workaround: After the print resumes simply smash the print head to one side with your hand, that will trigger crash detection (unless you're in stealth mode ofc.) and rehome X and Y.

shavenne, 2022-01-24:
I just had a layer shift caused by a blob in the middle of the print of my MK2.5.
This is one situation I wished I had a possibility to pause and just rehome X/Y.

So I searched and found this request here. As MK2.5 has no crash detection is there a way to get this function there?

3d-gussner, 2022-01-24:
@shavenne Sorry no, the MK2.5/S have the minirambo controller board. The crash detection and some other features are only supported on controller boards with TMC drivers, like the Einsy in the MK3/s/+ or the xbuddy board on the Mini.

leptun, 2022-01-24:
@3d-gussner that’s not the request. They want a manual rehome button during print on the lcd. Just like a “crash detected”, but on the lcd.

Prusa-Support, 2022-06-02:
Thank you for sharing your ideas.

I personally find this feature interesting and quite useful but I'd like you to review the scenarios where it will be most helpful.

If I understand correctly the most typical scenarios of use for such a feature are the following.
 1. Layer-shift due to a loose belt/pulley.
 2. Layer-shift due to the user accidentally hitting the extruder/heatbed in the same axis' movement direction (otherwise the Crash Detection is triggered).
 3. Crash Detection not triggered because of Stealth Mode (Crash Detection disabled).
 4. Layer-shift on an obsolete printer model (MK2/MK2.5 series).
 
In case n.1, continuing the print is not recommended. The print should preferably be stopped to run proper troubleshooting, as this could be concerning the HW side. That's totally understandable though that the user may take advantage of an "XY-homing" option to give it another chance.

In all cases, such a backup option to restore the print could be convenient but may be of little help if the layer-shift is not caught in a timely manner. In fact, a few shifted layers could be enough to compromise the entire print.
Also, as a backup opportunity, the user can cut the shifted layers and print only the missing upper print (cut in Prusa Slicer) to be glued on top.

Moreover, considering the "accidental nature" of these cases, probably only a mere fraction of these cases could be noticed in time (or predicted in advance) to properly restore the print on the go.

Last but not least, if an MK3- user does notice the issue (or is able to foresee it), a backup option could intentionally triggering the Crash Detection:
- switch to Normal Power mode (if not already enabled);
- sharply grasp a smooth rod as close as possible to the carriage, preferably on X-axis from both sides.
Of course, this is a viable option but not a real solution and not recommended.

This said, the feature could be very helpful but please do all due consideration and feel free to share your thoughts.

Michele Moramarco
Prusa Research

ulope, 2022-06-02:
@Prusa-Support you missed a case. Please see: #3460
Of course a proper fix for that issue would be preferrable, but until that is the case a manual re-home would be very helpful.

vintagepc, 2022-06-02:
In my situation the use case has almost always been when I've accidentally bumped the extruder position while e.g getting a mid print jam unclogged (MMU or otherwise). Note I'm also running my "jam detection" mod on my MMU printer which triggers if the FINDA reports filament but the IR sensor changes back to 0.

Prusa-Support, 2022-06-04:
> 1. Layer-shift due to a loose belt/pulley.
> 2. Layer-shift due to the user accidentally hitting the extruder/heatbed in the same axis' movement direction (otherwise the Crash Detection is triggered). [...]

Both cases could fall in the scenario n.2 but I should better say:

 2. Crash as a consequence of a generic incident, due to external factors.
 
Which, in your cases, are namely a tangled filament and an accidental hit.
These are only examples... The nature of the incident could vary a lot. The cause may even be the "infamous cat" playing with the cables 😄 and much more, but basically non-ordinary (supposedly rare) print scenarios.
Such a feature would be most helpful if the user happens to timely catch the problem but all the considerations about accidental events mentioned earlier can still apply.

Michele Moramarco
Prusa Research

warasilapm, 2022-07-03:
I just had a (very unusual) filament snag that I caught quickly but seems to have thrown off the position very slightly. This is a long (8+ hour) print, so ideally I'd be able to re-home the axes and continue with a small imperfection on my part. I'm giving it a go even without this feature since I really don't have the material on hand to start over, but having this feature would have been a great check to have just to be sure everything was still lined up.

It would also be nice to be able to home the z position in my case as well, since there was no material near the front-left corner where z homing is normally done.

Also @vintagepc hello from the old days of FFN! Good to see you out and about on the web!

wavexx, 2022-07-04:
Z rehoming in tune is tricky though.. it's very rare that it's possible to do except if paused within the first few mm of the print.
Or did I understand this wrong?

Prusa-Support, 2022-07-05:
Thank you for your feedback. Our developers could make use of this issue for future improvement of the firmware.

> [...] I'm giving it a go even without this feature since I really don't have the material on hand to start over, but having this feature would have been a great check to have just to be sure everything was still lined up.

For now, intentionally provoking a crash detection along the axis would trigger an XY-homing.

> It would also be nice to be able to home the z position in my case as well, since there was no material near the front-left corner where z homing is normally done.

Not only the implementation could be very tricky, but also the use. In fact, besides the front-left corner, the entire X and Y zone (left and front entire zones) need to be free, otherwise, the X-rods or the extruder-cable-wrap may collide.
There is a chance that Z axis was not offsetted, but that filament couldn't come out for a few layers. If so, it would not be possible to re-start the print from an earlier point of the g-code, it is not possible to re-print the empty layers.
In your case, my recommendation is to interrupt the print, sand the top, print only the missing upper part, and glued it to the top.

Michele Moramarco
Prusa Research

warasilapm, 2022-07-08:
Yeah z-homing in these cases is definitely more of a dream than a reality.

PR #2474 (2023-09-13): Add X-Y home to tune menu if print is paused
Fixes #2161. 

Note: This is intended to be merged after #2390 since it depends on the Tune menu being available while the print is paused. 

There are a number of reasons one might need this feature, e.g. accidental slip while dealing with filament change/jammed filament, catching a layer shift quickly enough, etc. 

I tested it on my printer (apologies in advance, I'm sure the forced X shift will make some people cringe...)

https://vimeo.com/390095133

and it homed and resumed flawlessly with only a tiny zit to show for it, about the same as you' [... cut]
closed 2023-09-13 by 3d-gussner