prusa3d/Prusa-Firmware #2162: [Request] Extrude 2-3mm on filamanet change

mvasilakis, 2019-09-01:
I consistently get jams when I insert a filament change (non MMU) when the nozzle cools down and then reheats to extract filament. I've run several tests and this is 100% eliminated when a couple mm of filament are extruded prior to unload. Again this is only an issue when the printer has requested a filament change and the user has not intervened triggering the nozzle to cool.

This is with a MK3s and a properly spec'd ptfe in the hot end. It seems to be less of an issue when the PTFE has opened up after long use but even then I see jams about 50% of the time. I also believe that pulling out a hardened filament tip forces open the diameter of the hole in the hot end PTFE prematurely. Dipping the filament and softening the tip also solves this.

Please consider modifying the code of the change filament action to include a mm or two of extrusion to soften the filament tip and ensure proper extraction. It needs just enough of a dip in the hot end  to soften it up.

gysnst2, 2019-09-09:
i would like to see the extruder go all the way to the right or left before the filament change starts.  ive had a few failed prints because of the location of the filament change and the purge of the filament

michalxfanta, 2019-09-19:
Thank you for your idea, we will try it and consider its implementation.

wavexx, 2019-11-09:
See #1095

closed 2023-10-10 by github-actions[bot]