prusa3d/Prusa-Firmware #2163: [3.8.0 RC2] Second print is not starting properly after SD card change

RCastoral, 2019-09-02:
A first print from SD card is OK while printing with single filament. If I initiate second print (without restart) but after remove/reinsert SD card, the nozzle won't start heating after selecting filament source. The printer heats the bed only and waiting. 

An ugly situation occurs when I try to pause such print as the Z axis starts descending and hardly hit the bed.

michalxfanta, 2019-09-19:
Is this problem persists with firmware 3.8.0 final?

RCastoral, 2019-09-20:
Yes, the problem is still existing in the final 3.8.0 firmware. However the way to deterministically reproduce it is a little bit tricky and I have spent few hours in last two days to find it out.

RCastoral, 2019-09-20:
So let's create following structure on SD card:
A1/B1/B1.gcode
A1/B2/B2.gcode

And then

1. Select to print from SD, navigate to B1.gcode and run it.
2. Stop the print.
3. Take out the SD card and enter it again. Click back to return to main menu.
4. Select to print from SD, navigate to B2.gcode and run it.
5. Stop the print.
6. Take out the SD card and enter it again. Click back to return to main menu.
7. Select to print from SD, navigate to B1.gcode again and run it.
8. Now the bed starts heating while nozzle is still set to 0.

It does not matter if you stop the print or wait till it is finished.

RCastoral, 2019-09-20:
[image]

cimoalpacino, 2019-09-21:
What's your start gcode script? Looks normal to me.
First it heats up the bed,. 
When temp is reached,it should start to heat the nozzle.
Am i missing something else?

leptun, 2019-09-21:
It is not normal. It should set both target temps and then wait for bed and after that for nozzle. Probably stopping the print messes things up. I’ll try to reproduce the issue and find a fix for upcoming 3.9.

cimoalpacino, 2019-09-21:
That's why asked for the start script.
I have changed it to heat up the bed first and only after that, i heat up the nozzle. So in my case showing 0 is normal 😊

leptun, 2019-09-21:
What is the reason for that. Is it to prevent oozing while starting or having a more uniform bed temperature?

cimoalpacino, 2019-09-21:
To prevent oozing and to have and even temperature before mesh bed leveling.
Before mesh bed leveling i heat the extruder to 170C. I heat it 8cm above the bed to reduce heating of the pinda probe. Only after mesh bed leveling  heat it up to final temperature. 
That's also helping to prevent oozing.

So my steps are:
1. Heat extruder to 170C (to soften any plastic on nozzle, so possible blobs don't affect auto home) 
2. Auto home xyz
3. Lift z by 80mm (reduce heat impact on pinda probe) 
4. Heat the bed to final temp. 
5. Heat extruder to final temp.
6. Do auto home with mesh bed leveling
7. Start printing

cimoalpacino, 2019-09-21:
On another printer i use the reverse:

Heat bed to final temp. 
Lift extruder by 50cm from current position (usually i have smaller prints, so it's never too high) 
Heat extruder to final temp. 
Auto home with mesh bed leveling

RCastoral, 2019-09-21:
Back to the original topic ;-)
@leptun It seems that directory depth is related to the problem. Following SD card structure 
A1/A1.gcode
A1/A2.gcode
will not cause the problem.

leptun, 2019-09-29:
Sooo... I tried replicating the problem and it got bad. Really bad. While I was navigating to the files I saw that the print fan started spinning for no reason at a weird speed. Then I selected the files and was greeted with a "Static memory has been overwritten" error after continuously moving through folders, starting and stopping prints and removing and inserting the card. The good news is that I figured out what was causing the issue. When the SD card was inited (after it is inserted or when the firmware starts), it didn't clear the variable that said the current directory depth and resulted in writing outside the allocated space for the filename, root and workdir, hence the overwritten memory error. I'll submit a PR with the required change.

leptun, 2019-10-03:
@RCastoral @cimoalpacino A fix has been merged. Please test if the issue is gone and close the thread if it’s fixed. Thanks for reporting. :)

RCastoral, 2019-10-04:
@leptun Good job. The issue is gone on 3.8.1 RC1.

closed 2019-10-04 by RCastoral