prusa3d/Prusa-Firmware #2164: [3.8.0 RC2, 1.0.6] MK2.5S with MMU2S - MMU2 Extruder Motor is too hot

ra100, 2019-09-03:
It's enough when I just turn on the printer, MMU2 unit doesn't need to do anything, but it's Extruder motor (on the side, where is just one motor) temperature starts to rise. And during the print it's getting higher (I don't have exact temperture, I think it's ~50 °C, I can hold it only for a little while). And I think that during the print, it's loosing ability to press against filament, so it's not loading/unloading properly.

leptun, 2019-09-08:
Do you use MMU Normal or Stealth Mode? Stealth Mode reduces currents when idling, while Normal Mode doesn't.

ra100, 2019-09-19:
Still the same is happening with 3.8.0, seems it doesn't have negative effect on printing performance. As I have other issues with MMU2, which I don't think are related to this "overheat". For comparison, Z-axis motors are colder than this one on MMU2 unit

JohnnyDeer, 2021-01-27:
This issue is duplicate of #1757, I´m closing.

closed 2021-01-27 by JohnnyDeer