prusa3d/Prusa-Firmware #2165: Prevent "Anywhere Z-Height" Routine from Interrupting Critical Movement Sequence

sungdzda, 2019-09-03:
This is a serious issue, so please read carefully. Thank you for your attention in advance.

**Behaviour #1**: When encoder is long-pressed, the printer unconditionally enters z-height adjustment menu and allows user to move the z axis. This is a convenience feature.

**Behaviour #2**: When printing stops, a sequence is issued (with MMU2) to briefly purge the filament before quickly retracting to unload.

**However**, if the user enters the z-height menu (per #1) and adjusts z position **before** the sequence in #2 is finished, the following scenario occurs.

1. The printer purges the filament
2. Z axis slowly moves. may take tens of seconds
3. The printer quickly retracts the filament to unload
4. The molten PLA filament is stuck in the hotend forever and the user has to painfully disassemble the printhead to service the printer. (lol)

**Suggested Solution:** Prevent entering the z-adjustment during any critical movement sequence.

michalxfanta, 2019-10-07:
This issue is solved in 3.8.1 RC1, please test and close this issue.

AbelCS, 2019-10-23:
> This issue is solved in 3.8.1 RC1, please test and close this issue.

Is this issue fixed for all situations or only related to MMU2?

I'm asking because I've noticed another serious problem with this. If you hold the knob and enter Z height mode when printer is homing, the nozzle will hit the bed.

There is a video showing it: https://www.youtube.com/watch?v=RJEidHT7U1Q

closed 2023-10-10 by github-actions[bot]