prusa3d/Prusa-Firmware #2168: 3.7.2/1.0.6 MMU Resets Between Color Changes

gummerch, 2019-09-05:
Setup:

- MK3S w/Bondtech on 3.7.2
- MMU2S on 1.0.6
- Skinnydip post-processing script (only way I can get the tips serviceable)

Every 10-100 tool changes the MMU unit resets when a filament unload is initiated. Process is:

- Extruder moves to wipe tower and initiates unload as expected
- Unload sequence proceeds all the way through the ejection from the extruder. This is the point
  when the MMU is supposed to take over.
- MMU unit acts as though I've pressed the reset button on the side. Lights cycle and selector
  barrel homes.
- By the time the MMU has come back online the filament is stuck in the extruder as the printer
  does not detect an issue with the MMU and is still trying to feed filament, expecting the next to
  be loaded.

Any insight would be greatly appreciated.

leptun, 2019-09-06:
Is it possible that you have a damaged MMU power cable?

Also please post the gcode. I’d like to test this on my mmu

gysnst2, 2019-09-09:
the MMU has a mind of it own sometimes. also sometime it will reset and bend the filament that is in the selector already.  or the printer will just reset.

closed 2023-10-10 by github-actions[bot]