prusa3d/Prusa-Firmware #2173: Question: MK3S PSU_Delta Define

vertigo235, 2019-09-06:
https://github.com/prusa3d/Prusa-Firmware/blob/9a60dd259bb69b14cc821c9db81acb59bd69254b/Firmware/variants/1_75mm_MK3S-EINSy10a-E3Dv6full.h#L30

Can one of the devs explain why this define is here, it looks to have something to do with Z height.  I have the silver PSU, so this comment would indicate that I should comment out this define.

However, in the official release, there is only one HEX file , so it's not clear why this definition is here.

Is the official release with or without this define?

Thanks

leptun, 2019-09-06:
The define is there to silence the black psu while idling. It should allow disabling Z motors. It shouldn’t cause any problems on the silver psu.

vertigo235, 2019-09-06:
Thanks @leptun that makes sense based on my skim of the code.

closed 2019-09-06 by vertigo235