prusa3d/Prusa-Firmware #2177: New firmware update Nozzle diameter setup

hutje, 2019-09-07:
> Nozzle diameter setup
This an initial release of this function. We need to finish the coding on the PrusaSlicer side in order to fully enable it. By default, all Original Prusa i3 printers are shipped with the 0.4mm nozzle. In case you change the nozzle, you can set the correct diameter in the printer's menu (see Settings -> HW setup -> Nozzle d.). The printer will then compare the stored value of the nozzle diameter with parametres in the G-code. If there is a mismatch, you will be informed to prevent a print failure.

Will this feature also correct the first layer test, where the amount of filament is being calculated when using 0.2mm, 0.6mm or 0.8mm?

3d-gussner, 2019-09-07:
@hutje As far i can see it will not compensate the extra/less extrusion. If you activate it it will warn you or even prevent the print depending on your settings.

I think the best is to make your own 1st Layer calibration gcode and store it on the SD card if you need another filament type or nozzle size.

leptun, 2019-09-07:
This could be generated on the fly though. Might be worth looking into in the future.

3d-gussner, 2019-09-07:
@leptun As this 1st layer gcode uses some space, I am not a big fan of it. And making the code even bigger to support different nozzle size and filament types doesn't make lot of sense to me.

leptun, 2019-09-07:
I think it is already generated on the fly with formulas. Pretty sure it can be easily implemented to support a different variable for nozzle diameter. If @DRracer approves this idea, I’ll look into it.

DRracer, 2019-09-10:
This feature doesn't perform any corrections in the G-code - it is here to warn the user, that the G-code was sliced for a different nozzle diameter.
Implementing 1st layer calibration for other diameters of nozzles is an interesting idea. Of course it is only about changing material flow through the nozzle...

Kachidoki2807, 2019-09-26:
To avoid opening a new issue, since the title is generic enough, I will expose my issue here.
On one of my machines, I use a 0.80mm nozzle.
So this feature is useless to me and I have to disable the warning.
I would like to address two points:

**First:**
As far as I read in the code, the selected value is stored in EEPROM as an integer (250, 400 or 600). And it is even stored once more (but why?) as an enum (25, 40, 60).
Maybe a more flexible way to handle that is to let the user to specify "numerically" the nozzle diameter instead of a list. It could even potentially save a small amount of FLASH.

**Second:**
About the first layer cal, I agree that could be useful to adapt the process to the nozzle diameter, Or at least, warn the user, or prevent him to do a self-hosted first layer cal if the nozzle diameter isn't 0.4mm.
I agree that the actual method (enqueueing G-Code ASCII coded) is really FLASH consuming, and it is not an option to duplicate them. Adding a live generated "E" parameter for each "G1" seems out of the scope, as it is the slicer that compute these values.
But I like the proposal of @DRracer to change the flow, it imply to add only one G-Code at the start and the value is a simple surface computation :
Flow = (100%) * nozzle_diameter² / (0.4mm)².
Application example: For a nozzle of 0.8mm => 100 * (800 * 800) / (400 * 400) = 400%

FYI: 0.25mm = 39%, 0.4mm = 100%, 0.6mm = 225%, 0.8mm = 400% and so on.

clarge249, 2019-10-14:
Please add in different sizes. At least support all of the standard sizes from E3D

closed 2023-10-10 by github-actions[bot]