prusa3d/Prusa-Firmware #2178: [New Firmware 3.8.0] Language Selection does not work

glowfishDE, 2019-09-07:
Hi,

i downloaded the 3.8.0 .hex for the Mk3S on the public prusa website and flashed my printer successfully.

After flashing the firmware the printer asks me to select a language.
I picked German/Deutsch. The printer says "Copying prusa Language files" and then "sorting files" and then "Please select a language" :)

Basically it asks me for a language selection over and over again until i choose english. Then i can use my printer. However i want to use the german language pack.

leptun, 2019-09-07:
Are you sure the language part of the update was flashed successfully? The first part (firmware) is with screen message and the second part (languages) has no visual indicator except the fans are spinning during the splash screen.

Please try flashing the firmware with PrusaSlicer (it is required for languages) again and post the output log if it still fails.

One more thing to try is factory reset/eeprom clear.

glowfishDE, 2019-09-07:
Ahh .. I flashed with octoprint (firmware updater plugin). How can i get it done 100% in octoprint ?

leptun, 2019-09-07:
I don’t think you can... You need the moddified avrdude version bundled with prusaSlicer. I’ll look into this on Monday.

3d-gussner, 2019-09-07:
> Ahh .. I flashed with octoprint (firmware updater plugin). How can i get it done 100% in octoprint ?

@glowfishDE You can ask the author of the Firmware updater plug in if she/he is willing to add the Prusa Slicer specific avrdude procedure.

glowfishDE, 2019-09-07:
Closed due to no firmware issue.

closed 2019-09-07 by glowfishDE