prusa3d/Prusa-Firmware #2179: FW 3.8.0, sheet rename menu. Holding button to adjust Z breaks menu.

BeldrothTheGold, 2019-09-08:
Hello Team, 

Love the new Sheets menu. Thanks for all of your hard work!

Noticed this bug quite by accident. 

It seems the menu stops advancing to the next character if you hold the button rather than quick push it. Like if you are attempting to adjust Z while in the rename menu.

Hope it's a simple fix. 

V/R
Scott

BeldrothTheGold, 2019-09-25:
@Panayiotis-git, @DRracer, @leptun 

I have tested the change @Panayiotis-git recommended from https://github.com/prusa3d/Prusa-Firmware/issues/1196#issuecomment-439501656.

This does indeed fix the bug which locks up the UI. I highly suggest this is tested and committed if it doesn't cause any regressions. 

Also, some brief testing of long-press in other menus with the fix in place validates @DRracer's statement that long-press still causes some strange behavior in other settings menus. 

I will work on this some more and submit another pull request.

BeldrothTheGold, 2019-10-03:
Looks like @DRacer was able to implement an excellent solution.  I will close this issue now.

closed 2019-10-03 by BeldrothTheGold

DRracer, 2019-10-03:
@BeldrothTheGold thanks ;)