prusa3d/Prusa-Firmware #2180: M860 PINDA temperature wait gives no UI information

Sembazuru, 2019-09-08:
I'm one of the many who use the M860 command in my start gcode to wait for the PINDA to enter the calibrated temperature range before starting the mesh-level process. Currently (firmware 3.8.0 RC2) there is no indication on the LCD screen that the printer is doing anything. It looks like something has hung up. I have found that in the serial terminal (I use pronterface, but I suspect any serial terminal will show this) I can monitor the current temperature of the PINDA vs. the set point. I see in the pronterface terminal lines like:
`P:27.3/35`
where 27.3 is the process value and 35 is the set point.

Would it be possible that instead of showing the gcode filename (I'm printing from SD card, not sure what would be shown using other print spooling methods) on the bottom of the LCD a similar readout to what is sent over the serial connection is shown while waiting for the PINDA temperature?

Because of the slow temperature ramp, showing the tenth's of a degree in the process value is useful. In the winter my printing area is usually less than 20°C so it isn't uncommon for the PINDA temperature to stall before 30°C and without seeing the tenths of a degree I wouldn't be able to track slow temperature changes.

hoppke, 2019-09-11:
Seconding this. I know some people issue a M117 before M860 to leave a mark on the LCD, but being able to actually track pinda temp on the LCD whilst waiting for it to hit target temp would be awesome.

vintagepc, 2019-09-13:
Ditto, I think I'll take a stab at implementing this tonight

vintagepc, 2019-09-13:
So there's actually already some code for this in the firmware for no-pinda cases. Looks pretty easy to extend to also handle M860

closed 2023-10-10 by github-actions[bot]