prusa3d/Prusa-Firmware #2182: FW 3.8.0, MMU2s 1.0.6 - MMU Stealth Mode detecting low power.

MikeThomas123456, 2019-09-09:
After flashing to v3.8.0 with MMU set to "Stealth" mode, the MMU consistently re-homes the idler and then flashes all 5 LEDs Red & Green. When MMU is set to "Normal", this condition does not exist. Flash back to 3.7.2 this condition no longer exists with MMU is "Stealth". 

Tried re-flashing 1.0.6 to the MMU and performing the MMU factory reset. Same condition still exists.

vintagepc, 2019-09-10:
I've been encountering this too. I'll have to double-check the setting but I think my MMU is in power/normal mode.

mazls, 2019-09-12:
I have the same issue, when i turn the Printer off and on again it works, for a while...

rcclausen, 2019-10-03:
I have been having the same issue. It started with the MMU re-homing every once in a while, switching to Normal mode seemed to help with that print. 

I checked my connections after that print and everything looked good. So I tried another print, hoping that normal mode would help. But it just went down hill from there.

Interestingly it seemed that when it would switch filament only 1 spot over it was fine, but when it had to travel farther it would cause a re-home or worse all lights would flash indicating low power.

I eventually had to revert my firmware back to 3.7.2 which has so far been working without any issues.

vintagepc, 2019-10-03:
Despite being in normal mode it was getting worse and worse on my printer. PR exchanged the MMU board and a new one is on the way for me; we'll see if that fixes it; the fact it was happening more and more seems to suggest it's not a firmware thing.

rcclausen, 2019-10-03:
I am curious if a new board fixes it for you. My first thought was also hardware, since it was getting worse and worse for me as well. But it seemed strange to me that other people were experiencing the same issue as me on 3.8.0 around the same time.

So I am currently running a 3 day print (24 hours in) with my firmware downgraded to 3.7.2 (MMU 1.0.6). So far the issue hasn't presented itself, which makes me think it might be firmware based. I just downgraded so I think its too early to tell. I will post an update when/if my current print finishes.

rcclausen, 2019-10-12:
Quick update. The 3 day print finished. This was on 3.7.2 and MMU 1.0.6 Stealth mode. There was 1 low power issue about halfway through. Thankfully it occurred right when I got back to my office. It seemed suspicious, though, that the issue coincided with me arriving home and turning on a bunch of electronics in the same room. I then upgraded to 3.8.1-RC and did 2 small MMU prints. The second print the MMU was on normal and it had a low power detection once.

So I am not sure what the cause is. It could be hardware. When downgrading to 3.7.2 I think I rechecked connections, so maybe that was why the issue got better, or 3.7.2 could be less sensitive to it.

closed 2023-10-10 by github-actions[bot]