prusa3d/Prusa-Firmware #2189: How to build on Mac?

xoxota99, 2019-09-11:
Following the Linux build instructions gives 

```
readlink: illegal option -- f
usage: readlink [-n] [file ...]
Picked up JAVA_TOOL_OPTIONS:
Error: Could not find or load main class processing.app.Base
Caused by: java.lang.ClassNotFoundException: processing.app.Base
```

What are the prerequisites / dependencies for following the build instructions? Any Mac-specific instructions?

william14235, 2019-09-11:
To download the firmware or drivers on Mac I think you have to go on prusa3d.com

Moorviper, 2019-09-11:
you have to download the arduino ide 
then add:
https://raw.githubusercontent.com/prusa3d/Arduino_Boards/master/IDE_Board_Manager/package_prusa3d_index.json
in settings in additional boards ....
then you can build for the MK3
hit build and upload done

bgiot, 2019-09-13:
Two options: 
- Use Arduino IDE (follow Windows instructions) 
- use VSCode + Platform.io extension.
You can have a look at my fork of prusa firmware reposition in BMGCustom branch. I use VSCode and platform.io to build the firmware.

closed 2022-12-29 by wavexx