prusa3d/Prusa-Firmware #2191: motors of Y axis and MMU2(S) does not turn off and keeps being warm/hot

Mirarkitty, 2019-09-11:
Firmware 3.8.0rc2 (and probably older).

Two motors of the MMU2 and the Y-axis motors keep running after the print. This seems to be a waste of electricity to heating the room.

[image]
[image]
[image]
[image]

The X and Z motors do turn off.

leptun, 2019-09-11:
That is not the Y axis, that’s the Z axis (up/down). It is always on in order to prevent a sqewed Z axis (one side higher than the other).

As for the MMU motors, only the idler should be kept on always and maybe even the selector. The pulley motor could be disabled, however that is not a priority.

There are also other reports on github about the MMU motors. You can search for those as well. Hope this helps.

Mirarkitty, 2019-09-11:
You are absolutely right with the Y/Z.

However I think it could very well turn off after say an hour and go back to the state it has when shut off (so it recalibrates Z). Same with the MMU.

Mirarkitty, 2019-09-11:
And this is absolutely a low priority issue, there's about 500 issues that should have higher priority. It's still an issue though.

leptun, 2019-09-11:
It will be kept in mind. We’re kinda busy at the moment with the fratures for 3.9. Thanks for reporting🙂.

gudnimg, 2023-07-04:
As we've improved in 3.13.0 the MMU side greatly in this regard so the motors are turned off when not used. I think we can close this issue once 3.13.0 is released. => adding this issue to our milestone as a reminder to close this issue.

gudnimg, 2023-07-31:
Closing as this is now resolved with FW 3.13.0 and MMU FW 3.0.0 :)

closed 2023-07-31 by gudnimg