prusa3d/Prusa-Firmware #2197: Power panic doesn't work

william14235, 2019-09-11:
When I tried out power panic the printer just froze whee it was without going to a safe place. The print them stuck to the nozzle and it became a failed print

N0ciple, 2019-09-17:
I think power panic doesn't work if you use the red switch on the PSU. This is the expected behavior. To simulate power panic, you should remove the plug from the wall instead. Although the recommended way of doing it is to plug your printer on a multiplug with a switch and to use the multiplug switch.

rollcast-rmc, 2019-09-22:
I am having the same issue following an upgrade to firmware 3.8.0 on my MK3S (upgraded). @william14235 Can you confirm your firmware version? I have seen this issue repeatedly on long prints since the upgrade. I unplugged the power cable from the wall and plugged back in during a print to verify. The extruder moved away from the print as normal, reheated, then returned to the last print location and simply stopped in place.

leptun, 2020-01-26:
This issue will be fixed by #2437

Prusa-Support, 2022-06-14:
Hello! Sorry for intervening so late on this thread. Closing the issue as this has been solved in the meantime, of course in case of anything similar still occurring, do not hesitate to let me know via email info@prusa3d.com mentioning me, we'll look into it in even more detail in that case. 

Alessandro Pantaleo
Prusa Research

closed 2022-06-14 by Prusa-Support