prusa3d/Prusa-Firmware #2198: MK3s/MMU2 Unresponsive While displaying "MMU OK... Returning to Position"

mmcglumphy, 2019-09-12:
I have now experienced this issue twice. Both times it has resulted in a stuck printer and ultimately a failed print. I will attempt to recount the steps I took with the most recent occurrence, but some details may or may not be necessary to reproduce. 

This issue began as a result of a failed filament swap. In my latest case, it was the first swap of the print. For what ever reason, the MMU failed to extract slot 1. It clicked around a few times and tried to move to the right while still having the slot 1 filament loaded down to the extruder. (!?)

Once the slot move failed, it stopped all activity with the 4th slot light lit (I believe yellow), all while still having filament down to the extruder in slot 1. (Slot 4 didn't even have filament loaded.)

Looking down at the MK3, it was just sitting there with the standard Info screen, showing 1 > 2, but not necessarily doing anything (I don't think I let it sit for long). So on the MK3, using the menus, I paused the print. The extruder lifted up and back. I then tended to the MMU.

I was able to manually unload the filament but holding down on the center button, letting the MMU rewind about 10cm or so, and repeating about 4 times to manually extract slot 1.

Then, using the MK3 menu, I was able to move to slot 2 and load the filament. (At some point, I remember that the MK3 also said that the MMU needs attention, but I do not recall at what moment I saw this.) In the end, I was able to get the next color loaded in slot 2, have the MMU happy (I'm fairly sure the filament was also to the extruder but not wiped), and the MK3 appeared to resume. The extruder returned to it's original position above the wipe tower, while displaying the message "MMU OK... Returning to Position".

At this point the MK3 became completely unresponsive. It was still maintaining heated temperature for the extruder and bed. But the knob did nothing. It wouldn't continue the print, even though it was completely in a position to resume without issue. 

I had to hit the reset button and fail the print.

I am not sure if things got into a weird state as a result of the pause. (Perhaps still paused in the background, but the menu is waiting on another event.) But, as I said in the beginning I have now had this happen twice. (Photo attached of first instance.) Simply using the interfaces, I was able to somehow get the printer into a stuck state and could no longer continue.

(Latest Firmware on all, along with gcode created in Slic3r 2.1-rc2)

[image]

mmcglumphy, 2019-09-14:
Just wanted to follow up and report that I had a similar situation occur last night - a filament failed to extract from the extruder head. (I think I have that problem fixed now. I tightened the extruder idler tension a touch.) Except this time, I did not Pause the print as soon as it failed. I waited for a bit longer and simply skipped Pausing. I just manually extracted the filament by holding down the center button on the MMU until it was out. At that point I was able to continue. I saw the "MMU Ok" message again as the extruder moved to it's resume position, but it actually continued. 

I has a suspicion there may be a bug with Pausing while in the middle of fixing an MMU loading issue. Where as, once the loading issue is resolved, the printer moves to resume. But the print is still Paused, and the state of the UI prevents any further action to Un-Pause the print.

Unfortunately, like I said, it is only a suspicion. I have experienced the issue twice, accidentally, but I'm not sure I want to waste time and material trying to reproduce the issue consistently. I will definitely report back if I see it again, otherwise.

DanielSMT, 2019-11-23:
I've experienced the exact same issue a few times now - so a filament change error and then after correcting it on the MMU2s being shown the message saying "MMU OK. Resuming position..." but then nothing happens - there is no way I can see to get it to continue at all and forced to press the x to fail the print.

It had completed some prints fine before this but a few hours into a new print this failed on the filament change and then the whole print is ruined.

Did you ever find a way to solve it?

Im using MK3S firmware 3.8.1 and MMU2s version 1.0.6.

mmcglumphy, 2019-11-24:
No. I've simply become way more careful with how I interact with the MMU.

My recent prints have minimal color swaps. I've had very few problems since keeping it simple, so I think I'm just getting used to some of the quirks. 

That last firmware fixed a few things that I noticed, so hopeful this workflow is eventually addressed. I'm pretty sure this is something that can be consistently reproduced.

Thanks for commenting. The more people that report similar experiences, the better visibility their issue will be addressed.

unixunion, 2020-09-02:
I have this same issue in latest firmwares. 3 times now while trying to solve load issues the mmu gets stuck on 

MMU ok. Resuming position...

mobilla, 2021-02-28:
having this same issue regularly, had to remove a jam in the MMU2 unit. After clearing the jam, i get a green light when loading the filament, click the wheel on the printer, it loads all the way to the nozzle, then goes back to the 'paused' position, never resuming the print. Screen says "MMU OK. Resuming position..." the entire time.

reidrankin, 2021-03-02:
Same here. I haven't gotten my idler tension dialed in just right yet, and I have frequent failed filament swaps. I've noticed that the combined printer/MMU state machine exhibiting clearly broken behavior in several not-very-close-to-the-edge cases, this being one of them.

reidrankin, 2021-03-02:
(FWIW, I have found that issuing an `M602` via my OctoPi setup will resume the print, though the menu doesn't come back...)

ssill2, 2021-04-24:
Just had this same problem this morning.  about 35% into a 52 hour print.  Had to trash the whole print since the buttons wouldn't respond on the mmu or the main printer lcd.  Very frustrating.

unixunion, 2021-04-24:
I have switched tool E3D tool changer for my multi material prints. My MMU is in the bin. Prusa has stopped caring.

ssill2, 2021-04-24:
it's disappointing.  I've had my eye on the PaletteV3Pro.  I've not heard of the E3D tool changer,  I'll have to check it out! Thanks!

buzkill420, 2021-12-27:
Have had this twice now just trying to get thru the test gcode... Come on. Prusa

unixunion, 2021-12-27:
The MMU made me resent printing, I had to remove it to enjoy my prusa again. It felt so good to go back to auto loading single filaments.

gudnimg, 2023-07-31:
Closing as this is now resolved with FW 3.13.0 and MMU FW 3.0.0 :)

closed 2023-07-31 by gudnimg