prusa3d/Prusa-Firmware #2205: MMU2 extrudes wrong filament, making print unrecoverable

JasXSL, 2019-09-14:
[image]

I'll have to investigate further, but it sounded like something caused the selector to get stuck. Even after clearing out any blockage, the selector remains on the wrong filament. Pushing either button on the MMU will only extrude the transparent filament into the air. This causes the whole print to fail, and there's no way to recover it.

JasXSL, 2019-09-14:
Update: Investigated and it's not the selector, it's the idler. I think it's trying to rotate the wrong way as it makes two thunks. It happens randomly going from filament 5 to 3 in this print. Figured out I can open it up and move the idler by hand to rescue the print. I'll post the gcode once the print finishes.

leptun, 2019-09-15:
Try and see id there is excessive resistance when moving the idler unpowered. Also try using MMU normal mode. One last thing to check is that the idler is aligned on the motor’s shaft.

cryocide, 2019-11-10:
I'm seeing similar behavior in mine, but I only just turned on serial logging in Octoprint to confirm which filament it should be loading vs. what the position is. In the most recent occurrence, the selector was on filament 1, the idler was feeding from filament 2, like something went out-of-sync. This was while printing the [basic MMU2 tester](https://www.prusaprinters.org/prints/1568-basic-multi-material-mmu2-tester ) so I think I'll be able to reproduce it.

jventerprises, 2020-03-09:
Hi,  

Any further progress on this issue?  I have one printer that is exhibiting this behavior as well

gudnimg, 2023-08-02:
Closing issue as this should be resolved with MMU FW 3.0.0 and Printer FW 3.13.0.

If there is any issues like this seen today it would be best to open a new Github issue as we don't want to debug the old firmware.

closed 2023-08-02 by gudnimg