prusa3d/Prusa-Firmware #2208: Disable MMU when setting up printer

Bob6162, 2019-09-15:
I have seen it recommended that when first turning on the MMU it would be better to make sure the printer is working and set up. That makes a lot of sense to me and it is what I tried without success. I even flashed my firmware so the printer would looks like a MK2.5S but it still wanted to load filament through the MMU. So, the suggestion is to add a switch (maybe in the Calibration menu) that turns the MMU off so the printer acts as just a printer. My guess is that you already have a switch in the firmware compiler that does that. This would make initial turn on and later troubleshooting much easier.

I ended up flashing MK2.5S firmware and disconnecting the MMU so I could set up the printer by itself but that isn't a very good solution

vintagepc, 2019-09-16:
Duplicate of #1876, #1698, and #2131

Bob6162, 2019-09-16:
Sorry for the duplicate. I tried to search but I guess it wasn't for the right key words.

maxcom99, 2020-02-26:
What is the current status of this, please? There are several more comments on the now-closed original request. https://github.com/prusa3d/Prusa-Firmware/issues/1876

closed 2021-01-21 by JohnnyDeer