prusa3d/Prusa-Firmware #2209: Z axis doesn't allow a negative value

Bob6162, 2019-09-15:
If I try to adjust the Z axis under Settings the “0” point seems to be set at the current Z axis location and a negative Z axis value can’t be entered. That is inconvenient in some cases (when troubleshooting). If I am trying to find the actual Z axis value for say the 4 corners of the heat bed. I can set one point manually then use settings and move X,Y, and Z but if one of the Z points is lower than the initial point it can’t be reached.

michalxfanta, 2019-09-17:
P.I.N.D.A must always be higher than the nozzle. And if you need to set the z-offset to a negative value, you need to move the P.I.N.D.A probe higher, further from the heatbed.

leptun, 2019-09-17:
He was refering to moving the z-axis down when position is unknown, not positive live-z.

This is risky since there is no way to prevent the user from crashing the nozzle into the bed if pinda is over the edge of the bed.

Bob6162, 2019-09-17:
I realize that there is a possibility of hitting the bed but the same possibility exist when you do a Live Z adjustment. It is certainly up to the user to insure that doesn't happen in both cases. However, for me at least, it would be an asset for troubleshooting. Having said that the same thing could be accomplished if diagnostic info is sent out the USB during X Y X cal. (But that is a separate suggestion)

Kachidoki2807, 2019-09-26:
After a reset, if the nozzle is really near of the bed, I like when the printer doesn't allow me to go down because I am turning the knob in the wrong direction once in two.
If I need to move down to the bed, I do an Auto-Home and then it is possible to touch the bed at Z0.
If really needed, a possible implementation could be a second long-press while being into the Z move menu that unlock the moving down. But I foresee a lot of issues regarding the management of the negative value.

I don't think this kind of feature is useful for a 3D printer, it is not a CNC.

Bob6162, 2019-09-26:
I agree with what you say in general. However when I am trying to troubleshoot a problem or setting up the printer the Z0 may actually be on a high point. If that happens I can't find a lower point if I can't control Z to a negative value.

tomatac, 2021-04-04:
Same here. I am stuck at the highest Z point and the printer thinks that that is zero. What do we do?

closed 2023-10-10 by github-actions[bot]