prusa3d/Prusa-Firmware #2212: Thermal runaway possible fix

Bob6162, 2019-09-15:
Thermal Runaway after upgrade to MK2.5S. I have watched as it prints and happens after the first lay when the fan turns on. I watched the temperature of the extruder as reported on the LCD and after the fan turns on the temperature starts to drop. It continues to drop until it is about 15 degrees below what it was programmed for then the Thermal Runaway error occurs. On the surface that seems to be what it should do under the conditions that I am seeing. So, that would lead me to believe that my extruder can't get hot enough. However, if the fan is disabled in Slicer it will print fine. When it fails I never see the fan speed decrease. Why not? The firmware can read the thermistor and can control the fan speed so when it sees that the temperature is dropping why doesn't it decrease the fan speed to allow the heater to bring the temperature back up? I have read a lot of different solutions for this Thermal Runaway issue but it seems to me that it should be solved in the firmware - not with bandaids.
I do have a silicone sock on my extruder and have looked at all of the suggestions that have been put out here in the Forum. The only thing I haven't tried was to change to the 40W heater as was suggested in one post. That might work but doesn't seem like it shouldn't be needed. I have been printing on my printer for a couple of years without issues. Since I did just do an upgrade I looked very critically at everything to make sure it wasn't caused by something I did. When I adjusted the fan speed in Slicer and that fixed the issue it seems clear that the heater is not at fault. I am guessing that new fan shroud is just more efficient and the heater can't keep up with the cooling.

zarathir, 2019-09-17:
Did you check the heater cartridge? I had the same issue and got it resolved with switching to a new heater cartridge.

Bob6162, 2019-09-18:
I don't rule that out but the current heater reaches high temperature easily if the fan is not on or turned down. I think the only purpose of that fan is to keep the extruder from overheating. In fact, it seems to be stopping the heater from reaching or maintaining the desired temperature.

tkircher, 2019-10-05:
I had a thermal runaway issue on my MK2.5S. I wiggled the hotend thermistor connector on the main board and it's been fine ever since.

ssvenn, 2019-11-17:
Same problem here with a brand new hotend assembly as part of the MK2.5S upgrade. I think a good solution could be a gcode command and menu setting to cap/scale the fan speed to avoid having to tweak it in the slicer.

I agree that the firmware should have a two-step approach to thermal runaway, trying to reduce fan speed for a few seconds to see if the temperature starts climbing instead of giving up immediately.

Aquilux, 2021-01-14:
Same here. I ran into this thermal runaway error twice now on the same print. Solution? I had the part fan set too high while bridging close to the bed, but that caused the firmware to panic. Maybe there should be either some compensation due to set fan speed, or at the least it should reduce fan speeds before it trips the emergency stop.