prusa3d/Prusa-Firmware #2214: [Feature Request] Detect Ground Filament

sarf2k4, 2019-09-16:
I had a lot of issues with filament jams couple of days ago with pla where the blob was too big to be pushed further down and requires extra force to pull it out of mmu2s body. After an attempt to use abs, turned out the nozzle clogged, changed to another cheap nozzle and I'm able to print with pla without jamming. It was a nozzle issues after all.

mk3s wtih mmu has 2 filament sensors, with my experience above, the printer didn't even pick up the filament being ground when it had jam or nozzle clog during printing. I did check the sensor status, IR = 0, FINDA = 1. 

I am requesting that this option will be implemented if IR = 0 and FINDA = 1, ground filament detected and initiate filament change and notify the user. But the sequence must have controls as the current versions of change filament with mmu2s, when it failed to unload the ground filament, basically nothing you can do except reset the printer and start over. This is because the extruder motor didn't move. In order to get rid of the filament jam we had to extrude a bit of filament to soften the blob and then it can be pulled out. 

I believe mmu2s can help the extruder motor's side of ground filament portion to let the extruder motor able to grip on a good side. 

Best if there's a control to adjust temperature and move both mmu2s and extruder motor in sync mode, same speed rather than a full lock down

michalxfanta, 2019-09-17:
Thank you for the idea, we will consider its implementation.

michalxfanta, 2019-10-11:
Thank you for your interesting idea, we will add it to our backlog.

closed 2019-10-11 by michalxfanta