prusa3d/Prusa-Firmware #2215: [3.8.0/1.0.6] MMU2S stepper motors not responding to MMU until power cut, reset does not work

Auge103, 2019-09-16:
**Initial situation:** 
- Hardware: MK3s with MMU2s
- MMU set to stealth mode in settings.
- MMU has filament paths calibrated about 20cm longer than default.
- MMU was used just for filament selection before a print. Print was about 12h long. 
After the print finished, the printer started to unload the filament, filament got stuck inside the extruder right above the filament sensor, triggering an MMU unload error. Nothing out of the ordinary so far.
- The printer displayed "MMU needs user attention". The MMU's first LED was flashing red.

This is where the bug starts:      
I fixed the filament blockage and then pressed the middle button of the MMU to try another unload. Nothing happened, but the LED stopped flashing, went red for ~10-30 seconds, then started flashing again. Huh, thats new?      
I then tried the same procedure three times, no motion of the MMU. 
I then reset the MMU via the reset button.
Still no MMU motor movement. No homing, no calibration, total silence. Still the LEDs indicated that the MMU calibrated and restarted. Strange.
The MMU went back to a single red flashing LED. I still couldn't unload the filament, no motor movement.

Annoyed by the MMUs missing response, I manually pulled the filament from the filament buffer back into the MMU until it was right at the gears and the Finda probe lit up again.
I then resetted the MMU another time. Still nothing. It indicated calibration, no movement.

_Now for the most strange part: When the first LED turned green, I tried to select filament#2. The MMU did a selector calibration without moving the selector. Then the second LED turned green, indicating a successful selector change. I was quite baffled. It was possible to switch between all 5 slots without anything moving at all. Thats the weirdest MMU bug I've ever encountered._

The only thing that finally got life into the MMU again was to cut the entire power supply and get the printer back online. No MMU or printer reset helped with that issue.

So with these kind of issues, operating an MMU is not fun anymore...

michalxfanta, 2019-09-17:
We prepare updated communication between MMU and printer and new control in error situations, witch will be solving problems like this.

CharlesWongNT, 2020-07-08:
Hello! May I know if this issue have ever been resolved?
I got the exact same problem since upgraded to 3.9.0 / 1.0.6 firmware for my MK3S/MMU2S.
All of the MMU motors eventually stop responding after a certain amount of time. The only way to return to normal is shut down the printer and turn on again (Reset on MMU or reset on front panel don't work). It is frustrating me for a month...

kbx81, 2020-07-30:
I've also seen this issue quite a few times; it happened again just today with the most recent firmware versions (3.9.0 / 1.0.6) and with prior versions, as well. I thought perhaps the MMU was overheating and shutting down as a result, as I have mine in a LACK enclosure and it seems to only happen after longer prints (3+ hours) and with higher temperatures (such as when printing ABS where the inside of the enclosure is quite a bit warmer). I see the same symptoms as described in the issue above; the MMU becomes more or less totally unresponsive and the only way to revive it is with a power cycle. It is not a particularly common occurrence, but it happens often enough that I took the time to Google it. 😄
I'm happy to troubleshoot further and/or to provide any other information if it's useful.

Auge103, 2020-09-21:
@kbx81 , @CharlesWongNT 
Well for anybody looking for a solution a year after I posted this issue, the cause of this whole thing was a connected Octoprint server, even when printing from an SD card. It seems that Octoprint will modify the Gcode even when printing entirely from SD.
To fix it, the user has to set up the printer in the Octoprint settings as "5 extruders, shared nozzle". 

Failing to to so will result in unpredictable behaviour of the MMU unit and the printer itself.
Examples of this are: Selecting wrong filament, moving the selector with filament already in the extruder, random printer resets, constant loading/unloading, MMU unit LEDs flashing, random MMU resets and so on.

closed 2020-09-21 by Auge103

kbx81, 2020-09-21:
Yes, I'm well aware of the nozzle config in OctoPrint -- I ran into other issues caused by this so I set it up very shortly after I built and connected the MMU2:
[image]
...yet I still have this behavior. I'm not sure I'd call this closed.