prusa3d/Prusa-Firmware #2222: Firmware 3.8.0 Smooth Sheet issue

Zarkex, 2019-09-20:
The Z Offset from 3.7.2 will be saved as Smooth1 even if you don't have the Z Offset Value of the Smooth Sheet set. This causes issues when you start the First Layer Cal. because it will dig into the bed if you install the smooth sheet. Maybe it should ask if the Value of 3.7.2 is for the Powder coated one or smooth one to prevent damage to the smooth sheet.

epicDISZ, 2019-09-22:
Yeah seems from the logs that it just saves to the first one, I would just change the name of the first sheet in the meanwhile. Hopefully they change it so it asks you which one to save to.

zoltan-l, 2019-10-13:
I have encountered the same after flashing the 3.8.0 now I have a sheet with nice holes in the starting point of the sheet.

Zarkex, 2019-10-14:
It's fixed in in the latest firmware. (Not beta)

closed 2020-03-17 by Zarkex