prusa3d/Prusa-Firmware #2224: nozzletemperature during filamentchange

Printer123, 2019-09-21:
setting: i3Mk3S (fw 3.8.0) & MMU2S (fw 1.0.6) 
The temperature of the nozzle during a filament-change changes as soon as the printhead reaches the wipingtower to the "new" print-temperature and the unloading of the "old" filament happens at the temperature of the "new" filament. 
E.g. when changing from BVOH (200°C) to PETG (230°C). Printing with BVOH happens  at 200°C, as soon as the printhead moves to the wiping tower, the temperature changes to 230°C and the unloading of BVOH takes place at 230°C. This leads to long tips and sometimes these long tips are blocking the selector of the MMU2 or are tangled in the extrudergear. Changing from PETG to BVOH happens the opposite way. PETG is unoladed at 200°C which is to low and can lead to a blocked extruder gear.
My proposition: The change to the "new" temperature does not happen until the "old" filament is completely unloaded (FINDA). So the "old" filament is unloaded at the appropriate temperature and the "new" filament is loaded at the corresponding temperature too. 
[image]
[image]

leptun, 2019-09-21:
This part of the unloading is handled entirely through gcode. There is nothing to change in the firmware since it only does what the slicer tells it to do. Therefore I suggest you post an issue there instead. https://github.com/prusa3d/PrusaSlicer

rtyr, 2020-08-31:
https://github.com/prusa3d/PrusaSlicer/issues/2971

closed 2023-10-10 by github-actions[bot]