prusa3d/Prusa-Firmware #2225: Modifying G28 mesh bed leveling Z stepover height

MandEdesign, 2019-09-21:
Hello Prusa folks, 

I am looking to modify the Z step over height for mesh bed leveling in order to clear a jig that I have permanently affixed to the build plate.   Would you be able to point me to the parameter or #define that I should be changing?   Your help is very much appreciated.

SnowMB, 2019-09-24:
have a look here: https://github.com/prusa3d/Prusa-Firmware/blob/ddc9cf6336bcd8e4fdd39061268ccdba8977089e/Firmware/Configuration.h#L353

MandEdesign, 2019-09-24:
Hi @SnowMB,   

Thanks for pointing that define out, I did try changing both of those values to 50 but it did not seem to change the stepover during mesh bed leveling.  

```
 #define Z_RAISE_BEFORE_PROBING 50
 #define Z_RAISE_BETWEEN_PROBINGS 50
```

MandEdesign, 2019-09-30:
I found another #define in the Configuration_Prusa.h file 

`427 - #define MESH_HOME_Z_SEARCH 15` which I changed to 15mm from 5mm(default) 

This change seemed to effect (M45 W) which is the Calibrate_Z function from the menu system.  I have been using this to clear the jig that I have however not really sure if it is leveling the bed in any meaningful way.

closed 2023-10-10 by github-actions[bot]