prusa3d/Prusa-Firmware #2228: Issue: Inconsistent language files, scripts results and translation files

3d-gussner, 2019-09-22:
The `lang_en.txt` and most `lang_en.xx.txt` end with a `LF` only line and not like `lang_en_fr.txt` which ends with a `<SPACE>LF` line.

This causes an issue that the `lang-export.sh` script does NOT add the last MESSAGE to the `/lang/po/Firmware.pot` and `/lang/po/Firmware_xx.po` files.
See [here](https://github.com/prusa3d/Prusa-Firmware/pull/2227/files#diff-a3ecf4c36356652cd76959b92a41921f) at the bottom of the file compared to [lang_en.txt](https://github.com/3d-gussner/Prusa-Firmware/blob/patch-5/lang/lang_en.txt) 

Only the French `/lang/po/Firmware_fr.po` contains all messages. 
See [here](https://github.com/prusa3d/Prusa-Firmware/pull/2227/files#diff-5f076d01f997ebf1c224ad0f28c24a5e) at the bottom of the file compared to [lang_en_fr.txt](https://github.com/3d-gussner/Prusa-Firmware/blob/patch-5/lang/lang_en_fr.txt) 
This will cause several issues:
- Some translators don't get the last/all messages to translate
- How ever the `lang-import.sh` will generate the `lang_en_xx.txt` files with the correct amount of messages.
  
So the question is:
1. Is there a bug in the `/lang/lang-export.sh` script?
2. Or should `/lang/lang_en.txt` and `/lang/lang_en_xx.txt` files end like the actual `/lang/lang_en_fr.txt` [file](https://github.com/3d-gussner/Prusa-Firmware/blob/patch-5/lang/lang_en_fr.txt)?

3d-gussner, 2019-09-23:
Update investigation:
- Checking the generated `/lang/po/new/lang_en_xx.txt` files by `/lang/lang-import.sh` script it seams that these are different from the `/lang/lang_en_xx.txt` files at this moment by ending with the `<last message/translastion>LF` and not having an additional line `LF` nor `<SPACE>LF`.

- Running `/lang/lang-add.sh lang_add.txt` shows that `lnag_en.txt` and `lang_en_xx.txt` files ending with **`<SPACE>LF` fail** but if they end with `<last message/translation>LF` or just `LF` it works fine.

Conclusion:
- What is the right format for the `/lang/lang_en.txt` and `/lang/lang_en_xx.txt` files? It is the format `lang-import.sh` script generates which means the `lang_en.txt` and `lang_en_xx.txt` MUST end with **`<last message/translation>LF`** and NEVER with `<SPACE>LF` or something else.
- `/lang/lang-export.sh` has a bug not exporting the last message to be translated.

DRracer, 2019-09-24:
@3d-gussner all right, let's add your findings into translations.md and fix the remaining inconsistencies.

3d-gussner, 2019-09-24:
@DRracer Another issue is that of the actual 360 messages ~192 have `No display definition on line xxxx` while running `lang-check.py` and depending on the language and 10-20 have `Text .... is longer then definiton on line xxxx`.

Should I try to update the messages in the code and translation files to minimize the Warnings and Error? As this is a time consuming task I would like to know your/Prusas opinion about it.

- This would be very helpful for the future as it would point out new code missing the `c=xx r=yy` documentation.
- Is there a limit of rows we can use for the long wizard messages? Often translations need more rows as the messages have to be split before they reach the 20 char limit of the LCD screen.
- Would it make sense to split words (if allowed in that language) to more readable way on the LCD screen.
  - In German you are allowed to split words by using `-` which is often used in news papers to fit the layout but keep the amount of chars in a line kind of consistent and makes it more "readable"/"nicer".

The `lang-check.py` script just checks if the translation has more characters than `cols*rows` but that isn't correct as the messages will be split before if the last word in row exceeds the 20 char length.
- Example:
  
  ```
  #MSG_FOLLOW_Z_CALIBRATION_FLOW c=20 r=4
  "There is still a need to make Z calibration. Please follow the manual, chapter First steps, section Calibration flow."
  "Es ist noch notwendig die Z-Kalibrierung auszufuehren. Bitte befolgen Sie das Handbuch, Kapitel Erste Schritte, Abschnitt Kalibrierablauf."
  ```
  The English text will be displayed on LCD in 7 rows to fit the max. 20 chars per row of the LCD

  ```
  There is still a
  need to make Z
  calibration. Please
  follow the manual,
  chapter First steps,
  section Calibration
  flow.
  ```
  and the German translation will be displayed in 9 rows

  ```
  Es ist noch
  notwendig die
  Z-Kalibrierung
  auszufuehren. Bitte
  befolgen Sie das
  Handbuch, Kapitel
  Erste Schritte,
  Abschnitt
  Kalibrierablauf.
  ```
  The 4 rows isn't correct even for English, but lets assume it would be c=20 r=7.
  The `lang-check.py` will only check the other languages exceed 140 chars (cols*rows).
  In this case the German translation is 138 chars long so it would meet the `lang-check.py` criteria BUT it doesn't represent the actual needed 9 LCD rows.

The MK2 branch `langtool.php` was more accurate than the MK3 branch `lang-check.py` script.

leptun, 2019-09-24:
I totally agree with you that the current cc*rr approach is problematic. While I was writing the first draft of the Romanian translation I made sure that line wrapping happened correctly and that everything fit nicely. It would definitely be helpful having a way to simulate all these long texts.

3d-gussner, 2019-09-24:
@DRracer @leptun Maybe we should add some code from the `MK2 langtool.php` into the `lang-check.py` script so the correct rows will be calculated. And more important check if the amount of translated messages is EQUAL to the english one. Having one more or one less generates real mess on the LCD screen, I have been there and had to factory rest my printer.

And it would be much easier if we had all messages in `messages.c/.h`. As mkbel pointed out in one of the last PRs that two `c=aa r=bb  c=xx r=yy` in one line are not allowed. Sorry that I mention again `messages.c/h` :-)

3d-gussner, 2019-10-03:
@DRracer Did anyone looked into the `lang-export.sh` issue?

closed 2023-10-10 by github-actions[bot]