prusa3d/Prusa-Firmware #2234: Question about MMU2S filament loading and GCode

vmario89, 2019-09-23:
Hi,
i have a question regarding to the following scenario which occured: i made a print with disabled spooljoin. the printer is controlled by repetier server. now the filament on spool ran out and the printer asks for filament loading action. But because i am not at the place physically i'd like to send some command to say "take filament spool 2". Is this possible? If yes, how? I checked the manual and i looked for some gcode reference. i did not find anything for this. How to remotely control the filament feed process without manual interaction at the printer itself.

regards, Mario

leptun, 2019-09-24:
At the moment it is not possible to do that due to multiple reasons:
- there is no such command implemented
- usb processing of commands is disabled while another action occurs such as a runout.

gudnimg, 2023-08-02:
Adding a comment that this is still relevant. I've seen similar issues that ask to somehow remotely select a filament when `TCodeChooseFromMenu()` is called. Currently we don't support it for technical reasons, if I recall correctly no G-code can be processed while the LCD is rendering a blocking screen.

closed 2023-10-10 by github-actions[bot]