prusa3d/Prusa-Firmware #2235: When running the 9 point calibration printer says "crash detected"

mvrueden, 2019-09-25:
I upgraded my Prusa i3 MK3s to the latest firmware, since then it is acting strangely.

A 6h print just died after 3h with the statement "crash detected. Do you want to continue".
As I was just out of the room for 10 Minutes I don't really know what happend, but said "yes".
But then it acted all strange, it went all the way to the right, even if it already was all the way up to the right, causing it to crash and push into the frame.

Rerunning the print caused the hotend to crash into the bed.
Autohoming did the same.

I decided to rerun the initial wizard.

However when I now start a print it does the 9 point reference check, however the middle one is not in the middle, but way off to the right. Besides that as soon as the printer wants to print the "filament junk line" it says again "crash detected" and does the whole procedure again, until it fails after a few times.

What is going on here?

leptun, 2019-10-06:
It is hard to tell what’s going on. It looks as mechanical problem to me. Does downgrading the firmware solve the issue. Also try reflashing and doing a factory reset.

mvrueden, 2019-10-07:
I tried all of the above and the issue still persisted. Loosened the x-axis belt a bit afterwards. Now it looks like it is solved. Haven't tried with the newest firmware though :)