prusa3d/Prusa-Firmware #2237: FW 3.8.0 printer starts moving when browsing files

AndWik, 2019-09-26:
This has never happened before but has happened on both my MK3 and MK3S with 3.8.0 FW.

Inserting the memory card with a new file. Navigating the memory card to find the file and suddenly the printer starts moving strange in both Y and Z direction.

The only, non standard, setting I use is sorting files by name rather than by date.

The memory cards differs in size and the number of files on it. Windows never reports an issue so the cards can be considered clean.

This may take some time to find. Hope you succeed!

Thanks
Anders
Sweden

leptun, 2019-10-06:
A fix has been merged and is available in 3.8.1-RC1. If the issue is solved for you, please close this thread.

AndWik, 2019-10-07:
Thanks a lot!!!

closed 2019-10-07 by AndWik