prusa3d/Prusa-Firmware #2238: first layer calibration for different nozzle sizes

ligniant, 2019-09-26:
I do have the following feature request:

Now it is possible to set the nozzle size in the firmware and you are implementing first layer calibration for different materials. It would be great if the built in first layer calibration could be valid for different nozzle diameters as well. Until now it is only usable with the 0.4 nozzle. Now it could be possible, that the printer is performing individual gcodes depending on the configured nozzle.

Kachidoki2807, 2019-09-26:
Duplicate of #2177

midnightsmith89, 2020-12-08:
OK, so this has been "scheduled" since 2019, whats up?  Had to search this tonight when trying to set up my 0.6 nozzle that is pre setup in the slicer! C'mon prusa!

closed 2023-07-24 by gudnimg