prusa3d/Prusa-Firmware #2241: BED PREHEAT ERROR - MK3 firmware 3.8.0

BenRosey, 2019-09-30:
Originally posted here: https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-hardware-firmware-and-software-help

MK3

Ever since firmware 3.5.0 I experienced the very annoying "BED PREHEAT ERROR" which shuts down the printer and makes it impossible to print.  I previously found a few articles and downgraded to firmware 3.4.1, completely solving the problem throughout the entire year.

https://github.com/prusa3d/Prusa-Firmware/issues/1362
https://github.com/prusa3d/Prusa-Firmware/issues/1435

A few days ago, I upgraded to firmware 3.8.0 because the "bed preheat error" was supposedly fixed in version 3.7.1.  Sure enough, immediately I got the same "BED PREHEAT ERROR" after upgrading to 3.8.0 on multiple printing attempts.

I went back to firmware 3.4.1 and happy printing again.

I am therefore lead to believe this is not a fixed issue in the firmware and needs more attention.

The ongoing error prohibits access to new features and OctoPrint is not able to smoothly print without the "you need firmware upgrade" notice that needs acknowledged every time.

michalxfanta, 2019-10-07:
Hello, we will find out what has changed since fw 3.4.1 and what causes this error. What material do you use for printing?

BenRosey, 2019-10-07:
PLA, ABS, and PET

 

Happens with everything.

 

Thank you.

 

Ben

Nippylicious, 2019-10-14:
Same problem with mk2.5s here too.

Since i updated from 3.4.1 to any further Updates, i got this error.

My workaround:
Example: when i Start my printer, my bed is at 20°c.

Then i hat up the bed to 25c.
Then to 35.
From here to the final Temperature without any error.
Seems that s safety timer reacts to quick.

Also, the first rising degrees are hardest reach, Once the bed is a little heated up, Then it goes quick.

mwhincup, 2020-02-10:
Since I upgraded to firmware 3.8.1 my initial preheat is successful, but after a couple of minutes of printing (ABS, with bed at 100degC) I get a BED PREHEAT ERROR.

closed 2023-10-10 by github-actions[bot]